CTCaer/hekate #5: Hekate, Fusee Gelee Payload for 3.0.2.

YellowCharmander, 2018-06-10:
Hi I just tried sleep mode with some old payloads I had and this one worked!
Although you need to disable screen lock in Settings, I'm not sure if you are aloud to select a Profile first because I didn't select a profile,
but you can enter the home menu by tapping the screen after launching this Payload with Hekate V2.0 (New Horizons) on your SD.
and enter sleep mode once (You have to enter sleep mode at least once in the home first) you can then wake up from sleep mode 
(using the power button to enter sleep mode and wake up only), also you can enter the Homebrew Channel and sleep and wake up.
I tested it, with my Switch on firmware 3.0.2.
Question: Do you think you can update your payload with this one for 3.x support and fix sleep mode with home button or power off and restart or with auto sleep mode?

[payload 3.0.2.zip](https://github.com/CTCaer/hekate/files/2088026/payload.3.0.2.zip)

YellowCharmander, 2018-06-10:
I am really sorry, that was the wrong payload.
Here is the actual sleep mode fixed for 3.0.2.
[payload 3.0.2.zip](https://github.com/CTCaer/hekate/files/2088056/payload.3.0.2.zip)

CTCaer, 2018-06-10:
Where did you found this payload?

CTCaer, 2018-06-10:
Never mind, found out from where.
This payload has the same sleep mode config with v2.3.

CTCaer, 2018-06-10:
Please load the following payload:
https://github.com/nwert/hekate/files/2088119/iplctc_sdram_id.zip

Console Info -> Print fuse info.
Reply back with the two numbers you see on top. (Sdram ID: # - #)

YellowCharmander, 2018-06-10:
It says; Sdram ID: 0 - 0
Do you want me to dump them to the SD Card?

CTCaer, 2018-06-10:
No need.

YellowCharmander, 2018-06-10:
No need for what?

YellowCharmander, 2018-06-10:
oh dumping them to the SD...

YellowCharmander, 2018-06-10:
I'm not familiar with Sdram for the Switch, but I am wondering if you are going to update your Payload to Hekate (2.4) soon because I request that you can hold home and go into sleep mode and I am not sure if you can do that on 5.1.0 with Hekate installed, can you?

CTCaer, 2018-06-10:
You can't do that with any hekate.

Hekate does not touch Horizon, it's a bootloader (like fusee primary/secondary).

Atmosphere does.

YellowCharmander, 2018-06-10:
Okay, have you tried Atmosphere by Atmosphere-NX?
Because I wish I could install it but I don't know how to compile to...

CTCaer, 2018-06-10:
The loader.kip and sm.kip you load for CFW are from Atmosphere actually.
Full atmosphere (with all its modules) is not yet supported.

YellowCharmander, 2018-06-10:
Okay, thanks.

Are you going to add this to the repository?
Because I found it in a Fusee Gelee folder on my Laptop after I was downloading all the
Fusee Gelee Payloads I could Find with Sleep Mode Fixed and I don't think I can find out
where download site is.

CTCaer, 2018-06-10:
I told you I found out where you got it.
It has the same sleep mode fixes with v2.3

Also I don't think you understand how this works.
You can't take the .bin file and add it to your code.

YellowCharmander, 2018-06-10:
Oh okay, So can I use your v2.3 payload to use Hekate the same way with sleep mode?

I understand now.

CTCaer, 2018-06-10:
Yeah.
Try it, and if it does not work (exactly like the one you linked), tell me.

YellowCharmander, 2018-06-10:
I just tried it and it didn't work.
When I went to the home menu and pressed the power button tto go into sleep
mode, it wouldn't wake.

YellowCharmander, 2018-06-10:
Okay so this is what worked...
The one I downloaded called (2.3) didn't work with sleep mode at all and the linked
payload only works in the homebrew launcher for sleep mode and not the home menu.
I pressed POWER and the immediately POWER.

CTCaer, 2018-06-10:
I see. Try to leave payload (3.0.2) **for a minute** in sleep and check if you can wake it up.

YellowCharmander, 2018-06-10:
Hey, I just tried it and it goes to sleep for a second then goes into a black screen, in
the homebrew launcher and the home menu.
I have to hold POWER to get it to turn off.

CTCaer, 2018-06-10:
Sorry to inform you but you never had a working sleep mode :/

> If you press POWER and immediately POWER, that's not sleep mode. **It doesn't have the chance to enter it**. **It just turns off the screen and re-enables it**.

YellowCharmander, 2018-06-10:
Oh okay.
But I was not pressing POWER and immediately POWER.
I pressed it once and then it went into a black screen until I held POWER to turn it off.

CTCaer, 2018-06-10:
That's what I'm saying. If you just press POWER once to enter sleep mode, it will wake up after a little and show black screen.

YellowCharmander, 2018-06-10:
So that must mean, I don't have a working sleep mode, right?

YellowCharmander, 2018-06-10:
So I just tried sleep mode with hekate_ctcaer_2.3 and it worked, I guess I was wrong, it
does work on 3.0.2 with sleep mode, sorry for bugging you...
The only problem was I can't enter sleep mode in homebrew without entering
sleep mode on the home menu first.
My apology, even the one that I put in this Issue does not enter sleep mode while
directly going to the homebrew menu.

YellowCharmander, 2018-06-10:
I would like to thank you for updating your payload for ALL Firmware's, maybe you could update it so
you can enter sleep mode while playing in Homebrew for a couple of minutes, or so you can leave it in
Sleep Mode for as long as you like, because mine get's a black screen after a minute while in sleep mode.

I guess it's safe to close this thread... so you can, unless you have something you would like to say first.

CTCaer, 2018-06-10:
You will surely find out when support for 3.0.2 will come.
It will be announced.

closed 2018-06-10 by CTCaer