CTCaer/hekate #13: Fix sleep mode for 3.X.X

CTCaer, 2018-06-18:


michaelshiel, 2018-06-23:
Any ideas on this? 

I've looked at the original diff for sleep support, and it seems like everything is in place for these versions. 

Could the warmboot_base be wrong? I'm guessing if the patchset was wrong for these versions it wouldn't boot at all?

CTCaer, 2018-06-23:
Yeah. Also last time I checked, the address was correct.

Yes, it would hang at boot.
I don't have time currently to work on this. If I want to try some things, are you available (what version you have)?

michaelshiel, 2018-06-23:
I am definitely available to test stuff. My console is on 3.0.1, and I'm very interested in the sleep support so I can keep my fuses the way they are.

I've got a fair bit of embedded ARM C experience (STM32, not Tegra) as well, so if you don't have time for the code but you know conceptually what needs changing/doing I can help out with that also.

vgmoose, 2018-06-25:
I have a 3.0.0 console and am also interested in this and can do some testing.

CTCaer, 2018-06-25:
The more the merrier.

@michaelshiel 
You can start by REing your nxbootloader. There's the most important stuff for lp0 power state.

CTCaer, 2018-06-25:
@michaelshiel @vgmoose 
Anyone available please contact me via email. (Comms via discord is preferable.)

cspencer49519, 2018-06-25:
I'm also on 3.0.0 and can test.

CTCaer, 2018-06-25:
Email me

CTCaer, 2018-07-07:
Fixed internally.

Will be released when some other changes are ready.

CTCaer, 2018-07-07:
Are you sure about cross-matching/checking everything in dissasembly?
The current behavior matches Nvidia's behavior on coreboot.
https://github.com/MrChromebox/coreboot/blob/master/src/soc/nvidia/tegra210/sdram_lp0.c

michaelshiel, 2018-07-07:
Yeah my mistake, I was interpreting the offsets incorrectly.

tumGER, 2018-07-09:
I think you can close this @CTCaer - Nice work btw :D

CTCaer, 2018-07-09:
https://github.com/CTCaer/hekate/commit/7af531de77b5f0e5b23843e16f520078d6351d93

@tumGER 
Thanks ^^

closed 2018-07-09 by CTCaer