CTCaer/hekate #14: SD card works on horizon and computer, but not hekate?

intraum, 2018-06-19:
My SanDisk Ultra 128GB microSDXC works perfectly with the switch OFW, and my computer, but hekate 'failed to init SD card' (after a couple second pause) whenever i try to print SD info or start CFW etc. 

I've tried formatting it as fat32, and exfat, with a variety of different tools and operating systems, including the switch itself. 

Are there still SD cards that don't work properly with Hekate yet? 

(5.1.0 with exFAT update)

CTCaer, 2018-06-19:
Not one that I'm aware of (except microSDSC).

Failed to init SD card stops at HW initialization. It doesn't even reach FatFS for filesystem type to play a role.

Do you have any other preferable way to communicate? I may need to send you a couple of different test/debugging payloads.

intraum, 2018-06-19:
i'll email you from my email, or can use discord if you prefer

CTCaer, 2018-06-19:
So this seems either an incompatibility or undefined behavior.
Either way, some extra checks may fix this.

Marking it as a bug for now

CTCaer, 2018-07-24:
In the end this was because the connector for the SD card slot was not seated correctly.

The sd card was falling in 1bit wide bus, which is something that is not supported by hekate.

The support will not be added, so a user will know that something is wrong. That's because the bus speed in 1 bit mode is very slow and the hardware defect should be fixed.

closed 2018-07-24 by CTCaer