CTCaer/hekate #15: Do not clear archive bit on Nintendo folder

rajkosto, 2018-06-24:
Pretty self explanatory, horizon expects the archive bit to be set on nca folders, part files and savedata, so it'd be best not to clear it on Nintendo folder or anything inside.

CTCaer, 2018-06-24:
Last time I had them, switch could not load any file at all.
In album? No photo/video shows up.
In ncas? Games asking for update and game sizes show up as 0MB.
In save? Game list shows cloud icon to every digital purchase.
Everything? Shows message that the sd card is corrupt.

This was in 3.0.2 and the last time I checked it.
If this is fixed in later version, I would still like to keep it as a quick fix for old versions.

Except if there's really a reason to keep them. Horizon expecting them is not enough.
What are the checks and the possible paths it takes?

Additionally, the sd card is expected to be used, backed up and migrated in a PC (and commonly in Windows which manipulate this bit).

CTCaer, 2018-06-26:
@rajkosto 
Any insight on what horizon does?

> What are the checks and the possible paths it takes?

CTCaer, 2018-07-02:
https://github.com/CTCaer/hekate/issues/25

CTCaer, 2018-07-02:
@austin5623 
What is your switch system version?
Additionally, cart games with an update on sd card, also crashed?

DarkMatterCore, 2018-07-02:
I can confirm that the same problem experienced by @austin5623 happened to me. My Switch is on 5.1.0, using a 32 GB micro SD card with exFAT format. My only game card title (Super Mario Odyssey) also failed to launch. I wasn't able to reproduce any kind of crash/kernel panic, though; just some simple launch errors that didn't trigger a reboot.

I managed to fix the problem by removing all attributes from all files and folders, and then adding just the archive bit to everything. Something like this:

```
cd /D X:
attrib -A -R -S -H /S /D
attrib +A /S /D
```

DarkMatterCore, 2018-07-02:
@CTCaer I agree on removing the option altogether. If it works for some and doesn't for others, it isn't right to call it a fix in the first place. Even better if it gets you additional free space to play with.

A disclaimer + an explanation in user guides should be enough, at least from my perspective.

CTCaer, 2018-07-02:
I agree with "If it works for some and doesn't for others, it isn't right to call it a fix in the first place.", in the sense that this should have a specific name. (which will imply that you should not casually use it).

I really want to see why that happens though.
Gonna run it again and see if I have a game that fails.

Btw, these errors you saw, still create error reports. Can you check the numbers there?

CTCaer, 2018-07-02:
I just checked all my games. All.
12 carts and 38 digital. All worked. No error.
I also checked the BoTW mods and a new Mario odyssey mod with full Atmosphere, and they also didn't crash.

I hope that these errors don't come from injecting "backups" via LayeredFS or running other "backup" loading tools.
If that's the case, it should not be used with these. Simple as that.
And it's better to tell me if that's the case, so I can plan the future of this fix attribute tool and not lose my free time without a need.

DarkMatterCore, 2018-07-02:
If these errors are supposed to log some kind of info, then they aren't. They're a lot like the error screen that appears after dumping a game with gcdumptool and trying to launch the game card without ejecting and re-inserting it (a bug I'm currently looking into). That error screen doesn't log any info as well.

And no, I have never used LayeredFS. At all.

CTCaer, 2018-07-02:
I see. Hmm..
This tool had 2 original intentions.
1. Remove archive bit from all files in Nintendo folder, to aid the migration of files.
2. Fix the homebrew not showing in hbmenu.

I'll probably limit it to the 2nd one.

CTCaer, 2018-07-02:
Ok, like you, I added the archive bit to everything.
Got 2016-0247. Unable to access microSD.

Then removed all of them and added them to individual folders and files.
Result was games crashing (missing update files) or completely missing (cloud icon).
Or with another folder, all games asking to update or redownload...

I still don't understand how we have completely opposite experiences on that matter.

CTCaer, 2018-07-02:
Let's make everyone happy:
https://github.com/CTCaer/hekate/commit/d4731bb540e1aa1d0cbce7ce9da978b6d59682af

closed 2018-07-02 by CTCaer