CTCaer/hekate #18: Add option to permanent disable specific eMMC locking features

CTCaer, 2018-06-27:
These will not be listed here until implementation.

Johnny165, 2018-07-01:
When will this be implemented? :)

CTCaer, 2018-07-01:
As soon as possible.

Johnny165, 2018-07-01:
Thank you so much CTCaer! I really appreciate it! (I have SX PRO and I really want to get rid of this malware, thought SX PRO was a good buy for external payloads, I was mistaken...)

Kasukoi, 2018-07-01:
@Johnny165
1. TX are some of the most disreputable people out there. Their intention is only to make money.

2. The payload loader doesn't require a buy if I remember correctly...

Just stay away from people like Xecuter or Gateway

Johnny165, 2018-07-01:
Ik, that's why I'm not going to buy from TX ever again. I got the SX PRO thinking it was a great product for loading external payloads on the go, I was wrong, completely. (Gateway is for 3DS only and their team is completely dead)

Kasukoi, 2018-07-01:
@Johnny165 
At the moment you're better off with an Android phone to load payloads on the go.
(I know GW is for 3DS. Nonetheless, their intentions were pretty similar to TX's)

Johnny165, 2018-07-01:
@Kasukoi 
I don't have an Android, I'm hard modding my Switch instead :)

Johnny165, 2018-07-01:
@CTCaer What happens if you run TX's payload? Does it just automatically lock the eMMC?

a7mag3ddon, 2018-07-01:
Stolen atmosphere code ? Jeez guys..  if that were the case why is atmosphere not released.

a7mag3ddon, 2018-07-01:
Also you can't steal code that's freely available. At worst all they dis was not credit atmosphere coders.

mariogamer2, 2018-07-01:
atmo is under gpl v2.
Also this is off topic.