CTCaer/hekate #21: Sleepmode issues in 5.1.0

borntohonk, 2018-07-01:
I'm having sleepmode crashes whenever I use version 3.0, leaving it on with backlight and refusing to go back unless powering down completely.

Using freshly compiled atmosphere kips.
[Atmosphere]
secmon=modules/atmos/exosphere.bin
kip1=modules/atmos/loader.kip
kip1=modules/atmos/sm.kip
kip1=modules/atmos/fs_mitm.kip
logopath=modules/hbgatmos/bootlogo.bmp
atmosphere=1

[atmos.zip](https://github.com/CTCaer/hekate/files/2153202/atmos.zip)

GGLinnk, 2018-07-01:
It seems that sleep mode turn back switch into RCM Mode

GGLinnk, 2018-07-01:
Then it's possible to launch hekate but launching atmosphere a second time without reboot switch seems to freeze.

GGLinnk, 2018-07-01:
This issue it's not an hekate issue but an atmosphere issue.
No issue with CFW or LayeredFS.

CTCaer, 2018-07-02:
@borntohonk 
That's because Exosphere's deep sleep (LP0) functions are not ready yet.

SciresM said before that he's gonna look into that when possible.

closed 2018-07-02 by CTCaer