CTCaer/hekate #29: comment phrasing bug?

eggy, 2018-07-03:
When troubleshooting an LFS related crash, it appears there is a bug with comments..
Tested this with SMO on FW 5.1.0.
Here's my ini file (with the bug):

{-------- Stock -------}
[Stock]
logopath=cfw/custom.bmp

[Stock (SVC perms)]
logopath=cfw/custom.bmp
fullsvcperm=1
debugmode=1
{ }

{-- Custom Firmware --}
[AMS HBL]
kip1=cfw/local/loader.kip
kip1=cfw/local/sm.kip

[AMS HBL (SVC Perms)]
kip1=cfw/local/loader.kip
kip1=cfw/local/sm.kip
fullsvcperm=1

#[LFS (automatic)]
#kip1=cfw/local/loader.kip
#kip1=cfw/local/sm.kip
#kip1=cfw/local/fs_mitm.kip
#fullsvcperm=1
#debugmode=1
#atmosphere=1

[AMS (full)]
secmon=cfw/exosphere.bin
kip1=cfw/exosphere/loader.kip
kip1=cfw/exosphere/sm.kip
kip1=cfw/exosphere/fs_mitm.kip
#kip1=cfw/exosphere/pm.kip
logopath=cfw/exosphere/ams.bmp
atmosphere=1
{ }

In the above example, the LFS (Automatic) entry doesn't show, as expected as its commented out.
However, if you try to load AMS Full, LFS has an instant game crash implying something is off.. removing the commented out pm.kip line resolves the crash.. Looks like there's a bug with using comments in the middle of a boot entry..

CTCaer, 2018-07-03:
There's no bug.
You deviated from the template.

The comment is recognized as an entry. Because you have it in the middle of a boot entry's configuration, you split it in half and the logopath and atmosphere configs are ignored.

Additionally, because these 2 now fall out of a boot entry, with the next config edit inside hekate, they are removed from your config.

So you have to open again your config and change it to
```
[AMS (full)]
secmon=cfw/exosphere.bin
kip1=cfw/exosphere/loader.kip
kip1=cfw/exosphere/sm.kip
kip1=cfw/exosphere/fs_mitm.kip
logopath=cfw/exosphere/ams.bmp
atmosphere=1
#kip1=cfw/exosphere/pm.kip
```

So, when you comment out config lines, you have to also put them last in the order.

eggy, 2018-07-03:
I made an assumption on how comments work, I assumed if I commented a line in an entry it would be skipped and the remaining lines after the comment would be considered part of the same boot entry. From this feedback it sounds like by design, it doesn't work this way.

CTCaer, 2018-07-03:
Yes normally that's how it works.

But here, I use the ini parsing to deconstruct and reconstruct the .ini file, so hekate can edit it and save any configuration done.

For this reason the comment (#) is considered an entry.
Another reason, is that if a non key string is found (without "=" ), it creates problems with booting.
Lastly, having more code to check, will end up fattening the payload.
So let's say that this is by design and by limitations.

I'll keep this issue open for now, as a reference.
Until more people understand the new .ini layout.

CTCaer, 2018-07-07:
I believe, most learned how this works.

closed 2018-07-07 by CTCaer