CTCaer/hekate #30: Verification forever bugged...

CTCaer, 2018-07-03:
There's a new issue with verification.

Currently it tries to parse a u64 number with a u32 storage.
This way it tries to compare the wrong eMMC segment with the correct SD one.

A fix is planned for release ASAP!
It's currently in testing, by users that had this problem.

cleggorz, 2018-07-03:
Is this the same error that results in

SD card and emmc data (@LBA 00000200) do not match.
Verification failed.

CTCaer, 2018-07-04:
The x2000 bug was only a false printing.
It has 3 bugs:
* For partial backup, it does not take the correct start LBA after continuing a partial backup
* For single file (exFAT) it uses a 32bit storage for a 64bit value
* Any of the 2 may break if it reaches the ~70 minute mark.

The last case is the only one not fixed internally. It's obscure and strange.

CTCaer, 2018-07-04:
Fixed (?!?) in:
https://github.com/CTCaer/hekate/commit/ebb9ca5bf56c28824cfbea4b0cb0668e6df7cb32
https://github.com/CTCaer/hekate/commit/5e8eb1c57aa5498d056482ecb808f0d96abd6bf2

Awaiting testers' results before closing.

cleggorz, 2018-07-04:
Testing now but this seems fixed. Already past where it would normally fail.

CTCaer, 2018-07-04:
I know.
Internally, it succeeded for all testers.
(And I currently test verification for restore option).

But I'll clean up the issues when the code reaches release state.

CTCaer, 2018-07-05:
All testers and users on v3.1 release report that it doesn't break anymore.

closed 2018-07-05 by CTCaer