CTCaer/hekate #31: Hoping to be able to fix or upgrade it over the Internet.

luhongtao, 2018-07-03:
Hope to bring a repair machine with a formatter when reinstalling the restore program, to ensure that the machine bad program empty to prevent it from being ban,3.0.0 repair failed rawnand.bin. is a backup of 00-14, black screen, but can enter rcm. hope to have a repair tool, thank you

luhongtao, 2018-07-03:
Inadvertently brush 5.1.0 system formatting, resulting in program confusion black screen can enter the rcm, restore to the system, do not know the format of the switch format, I hope to repair

CTCaer, 2018-07-04:
I still don't understand what you are saying.
(Try to write your problem in your language, I suppose simplified Chinese.)

CTCaer, 2018-07-05:
https://github.com/CTCaer/hekate/issues/34

closed 2018-07-05 by CTCaer