CTCaer/hekate #32: Boot Linux

specialkk, 2018-07-03:
Not sure if this is at all on the roadmap or even within the scope of this project but it would be pretty convenient to have one bootloader that can dual-boot Horizon and Linux. Ideally it would be able to read everything it needs from SD without needing coreboot to be piped over USB. Would it be as simple as having Hekate replace the role of CBFS described here https://fail0verflow.com/blog/2018/shofel2/#linux-on-switch-boot-chain?

Regardless it would be ideal to have some solution for internal hardmods like with the Trinket M0 that don't have the flash capacity for all the parts necessary to boot Linux. Also sorry if this sounds like a bunch of nonsense, I'm a bit new to some of this boot-level stuff.

CTCaer, 2018-07-04:
This can't be done.

But, it may be able to load coreboot though.

I'll have it in mind.

CTCaer, 2018-07-07:
Closing in favor of https://github.com/CTCaer/hekate/issues/20

Payload booting implies the support of linux booting.

closed 2018-07-07 by CTCaer