CTCaer/hekate #33: /switch folder missing from the archive option

plaidman, 2018-07-04:
I've compiled a new .bin from the most recent master branch to test the archive bit update. It doesn't seem to be working when I use the 'switch folder only' option. From what I can understand of your code, I think it might be opening the /switch folder to list all files and unsetting the archive bit for those files inside it doesn't actually unset the bit for the /switch folder itself.

I was able to verify by loading up a windows computer and unsetting the archive bit on only that folder fixed the issue for me. Homebrew works and all switch games work!

CTCaer, 2018-07-04:
Thank you.
Gonna check this today.

CTCaer, 2018-07-04:
Ok I saw the issue. We enter the directory right away without checking its object.

CTCaer, 2018-07-04:
fixed in https://github.com/CTCaer/hekate/commit/ebb9ca5bf56c28824cfbea4b0cb0668e6df7cb32#diff-799c7cc4c061751437d68637b5d9ae27R1895

closed 2018-07-04 by CTCaer