CTCaer/hekate #34: Is there no way to restore the machine?

luhongtao, 2018-07-05:
[image]
用的join_15_2GBparts_windows 是不是需要返厂了，黑屏进不去系统

luhongtao, 2018-07-05:
Join _ 15 _ 2GBparts _ windows need to return to the factory, black screen can not enter the system

CTCaer, 2018-07-05:
What you did was a restore of the one of the 3 physical partitions.
You need to also restore BOOT0 and BOOT1?
These are essential to do a FULL restore.

The black screen you are saying, is probably from incompatible rawnand.bin and the current BOOT0/1 you have.

So, copy the BOOT0 and BOOT1 backups you did, to /Backup/Restore and run the option "Restore eMMC BOOT0/1".

GGLinnk, 2018-07-05:
Rajkosto's tools can help you in nand restoring :
1. Use [gptrestore payload](https://files.sshnuke.net/gptrestorev1.zip) to fix gpt table.
2. With [HACDiskMount](https://files.sshnuke.net/HacDiskMount1044.zip) extract all parts of your rawnand.bin (Use your dumped biskeys : can be dumped by using [this payload](https://files.sshnuke.net/biskeydumpv6.zip) ).
3. Use [memloader](https://files.sshnuke.net/memloaderv1.zip) to put your switch into USB eMMC UMS Mode.
4. With [HACDiskMount](https://files.sshnuke.net/HacDiskMount1044.zip) with admin right, open your eMMC nand by opening device Linux UMS Device 0, and restore 1 by 1 extracted files from rawnand.bin by using biskey too.

luhongtao, 2018-07-06:
[image]Why filling in key fails

oCanna, 2018-07-06:
Can you start the recovery mode pressing power + vol+ + vol- ?

CTCaer, 2018-07-06:
@luhongtao 
Check that you got your BIS keys correctly.
This works great here.

luhongtao, 2018-07-06:
Can enter rcm. system 3.0.0 can not restore firmware, restore the end of the black screen, you can now view the biskey

CTCaer, 2018-07-06:
You tried to launch firmware through hekate?

luhongtao, 2018-07-06:
If you want to resume booting, you will not be running the system on the black screen at the moment.

CTCaer, 2018-07-07:
In the end, because this is not an actual issue with hekate and here is not a support forum, I'm closing this.

@luhongtao 
Try to get help through support forums.

closed 2018-07-07 by CTCaer