CTCaer/hekate #39: dump package1 twice in 1 run hangs init of SD card

DacoTaco, 2018-07-07:
when playing around with the SD mounting/device handling/FSFat i noticed that trying to dump package1 twice made it hang/crash. no error on SD mounting so it either hangs somewhere else

didn't have time to look yet, im gonna try and get my changes done first

CTCaer, 2018-07-07:
Hmm I'll check it.
It may be a memory leak or corruption.

DacoTaco, 2018-07-07:
if you wait long enough screen changes, this might be an indication that it is indeed a leak.
maybe the size is incorrect and it starts dumping where it shouldn't?

are you just printf'ing to screen btw when debugging? (easy to know what a good method is :P )

CTCaer, 2018-07-07:
I normally debug with screen printing. Sometimes with USB->PC printing.
(Depending how close the code is and if it can corrupt screen printing).

What you describe is the definition of memory corruption.
Memory leak is when you acquire memory, done with it and it can't be used later by anything else.

DacoTaco, 2018-07-07:
ye, so standard procedures in debugging.
had hoped on better debugging tools, but guess not for now :') (didn't follow the progress of switch tools at all xD )

ye ok, i confused leak & corruption for a sec. my mistake.

CTCaer, 2018-07-07:
The problem starts from keygen or before.
After that it wrongly decrypts pkg11

CTCaer, 2018-07-07:
OK, the problem is not mem corruption.

Keygen is supposed to only be used once, as it configures SE.

Additionally I found several mem leaks.

All these, are fixed internally.

DacoTaco, 2018-07-07:
sweet

CTCaer, 2018-07-07:
Fixed in https://github.com/CTCaer/hekate/commit/60905c3829f234a0ceecb0ace3dffb48482743f6

@DacoTaco 
Thank you for reporting that.
This made me also look for other leaks and I believe that the launch code is now completely free of memory leaks.

closed 2018-07-07 by CTCaer

DacoTaco, 2018-07-07:
np. just came across it during a little test of mine (getting decrypted nand access from RCM. im abusing hekate & FATFS for it xD )

CTCaer, 2018-07-07:
Keep in mind that there's a hard cap on payload size (126296 bytes).
(A fast way to get a smaller size is to comment CUSTOMDEFINES in makefile and change -O2 to -Os, while working on this)

DacoTaco, 2018-07-07:
ye i read about it. in the back of my head ive been thinking if maybe it isn't better to try and implement booting other payloads from hekate. but i know less about that then the nand & decrypting xD