CTCaer/hekate #41: Request - File Manager

lolicop123, 2018-07-07:
Although we have nx-shell, a File Manager built into Hekate would be cool.

br4z0rf, 2018-07-09:
Not OP but I figured it's related. I'd like something à la adb to pull and push files in the SD card so that we don't have to take it out every time.  
I could start working on a PR if you are interested.

CTCaer, 2018-07-09:
That's irrelevant with OP, yes, but an adb push/pull or even sideload would be very nice.

The problem is size constraints (~123KiB). So, I believe that payload launching would be the top priority.
If we can support that, then memloader is enough for all of these.

mariogamer2, 2018-07-11:
I think this is not worth it, even to select backup place, or something. (size...)

CTCaer, 2018-07-11:
A type of file browsing will be included for payload launching.
But a normal file manager? nope.

closed 2018-07-11 by CTCaer