CTCaer/hekate #42: A horizontal menu would be great

coffeeken, 2018-07-09:
If there is a horizontal menu, the experience will be much exhanced.

CTCaer, 2018-07-09:
If there was touch support yes. But touch support is no for hekate, because of size constraints (we need to upload the touchpanel's firmware in every boot).
Otherwise the UX is regressed by having to pressing the volume buttons in 90o degrees orientation.

Additionally, a horizontal menu, as much as I want to do it, it can degrade performance and also reach size limits.

chaoskagami, 2018-07-11:
Why would it degrade performance? All you need to do is calculate coordinates slightly different. You can dirty render over the previous screen instead of refreshing the whole display, and the only performance hit should be when scrolling occurs due to needing lines memcpys of size (yaw - bpp) rather than (lines-1)*yaw memcpys for the whole yaw. I haven't read the code, but I'm assuming you're working with a 720x1280 framebuffer. Even then, both are slow without hardware assistance. It shouldn't be that different. Just avoid scrolling.

People have to deal with the volume button mapping thing on many phone-like android devices and tablets. It's generally accepted that the left button is up while the right is down if the buttons are on the top of the device, and the reverse otherwise.

For touchscreen firmware - I don't think that's what this report is really about. It's about the inconvenience of the ninety degree angle. I personally would just require users to provide it for touchscreen to be used.

mologie, 2018-07-12:
Could you not load the touchscreen firmware from eMMC? Horizon should store it somewhere after all.

CTCaer, 2018-08-06:
Closing in favor of https://github.com/CTCaer/hekate/issues/65

closed 2018-08-06 by CTCaer