CTCaer/hekate #45: Re-enable the BQ24192 BATFET turn off function

oCanna, 2018-07-11:
It would be nice to have this fuction to disable the BATFET.

As probably more and more people are going to install an internal modchip connecting it to the BQ24193's SYS pins (15&16), it would be really greate to be able to disconnect the power so the modchip can't drain the battery if you plan not using your console for an extended period of time, want to ship it or whatever.

Making sure that the battery is fully charged and there is an external power source connected to the console, like it was done in the reset_pmic_fuel_gauge_charger_config() function before turning off the BATFET, should also be (re)implemented.

CTCaer, 2018-07-11:
Add a switch to your modchip.

This _can_ create more problems than it fixes and also it resets all battery stats. Not good if you want to know your real %.

(You can still compile it and use it though)

closed 2018-07-11 by CTCaer

mariogamer2, 2018-07-11:
Little unrelated but isn't the new fix not going to make the battery less good because of the fact that other values are still bad?

CTCaer, 2018-07-11:
Based on some users, the values fixed themselves with only 1-3 full charge cycles.

This is also described in the datasheet. So the above sentence means that the fuel gauge does its job.