CTCaer/hekate #46: support for png/jpeg boot logos?

lolicop123, 2018-07-11:
This would be helpful to the people that cant get 32 bit bmps and will allow for more editing software to be used in creating boot logos. It would also make it easier for anyone to have a custom boot logo without needing to edit an image too much. it would be much easier than trying to get a 32bit bmp image

oCanna, 2018-07-11:
In my opinion it is easy enough for the average hekate user if he is willing to read a little bit.
It would be probably sufficient to mention in the bootlogo readme that all you have to do is using the latest gimp release. JPEG/PNG support would unnecessarily blow up the size of the payload if there even is any left for things like these.

CTCaer, 2018-07-11:
There are many editors that support quad channel BMPs.

Additionally, as said before, jpeg or png support will greatly increase the payload size (we are already at max) and it will kill the auto boot performance by needing around 5s to only show the bootlogo.

closed 2018-07-11 by CTCaer