CTCaer/hekate #47: Ability to restore from split backups (if all of them are together)

aveao, 2018-07-11:
So, recently I took a backup to a 16GB microSD card. It split them into 15 parts due to reasons, and when I wanted to restore them using a 64GB microSD card today (on exFAT ofc), I simply copied all 15 files to the microSD card, but just found out that hekate doesn't support just reading them all.

This would be pretty nice to have so that other people can avoid what I am doing right now (which is removing 32gb of backups from microSD, combining them and then copying again, wasting a couple hours in the process due to my laptop's lack of sdxc support).

If you want I can try to write support for this and send a PR.

CTCaer, 2018-07-11:
Yeah, this is not supported for now.
I wanted to make sure that restore is ok first.

But right now the top priority is payload launching which I want to implement since a couple of weeks.

emvym3two, 2019-02-25:
I have started writing a version of this since I wanted the same..

The version is working but I had to cut out the verification code with #if's for now since it makes this condition no longer true

> Max size is 126296 Bytes

I'm re-factoring the code a bit to try to cut down on the size by putting code which does the same thing into functions.

The tag for what I've done is here
https://github.com/emvym3two/hekate/tree/v4.6svcsplitprerelease

And a compiled release version of it is here.

https://github.com/emvym3two/hekate/releases/tag/v4.6svcsplitprerelease

I'll have to have a look at splitting it out into a module which can be loaded so that the size won't be an issue (especially since I noticed the main branch is now only a few bytes off what I'm assuming the size limitation is)

emvym3two, 2019-02-25:
> The version is working but I had to cut out the verification code with #if's for now since it makes this condition no longer true
> 
> > Max size is 126296 Bytes
> 

When I build the 4.8 tag it's actually saying

> Payload size is 127607
> Max size is 126296 Bytes.

Rather than the 126283 bytes you have managed.

Do you know what would be making the file size larger for this.
(Sorry c skills haven't been used in about 10 years)

CTCaer, 2019-02-25:
I used latest gcc which fixes a bug that was introduced around October or September and lowers the size again on -O2.
Btw if you want to work on it and PR, use the dev branch here.
And I can fix the sizes.

CTCaer, 2019-03-06:
@emvym3two any news on that?
Only this is left for a new release.

I can work on it if you don't have time.

fennectech, 2019-03-07:
This would be amazing.  Then we dont have to find a 32 gig card to format for a restore.

CTCaer, 2019-03-08:
https://github.com/CTCaer/hekate/commit/50eff413516134add07a6bc99dbfd23ba2e1be9b

Thanks @emvym3two

closed 2019-03-08 by CTCaer