CTCaer/hekate #49: SD card mount/init failure should not default to booting stock firmware

misson20000, 2018-07-18:
Currently, when the SD card cannot be initialized or mounted, a message is displayed on screen for a few seconds and the stock firmware will be booted. Every time this has happened to me, it's because I've forgotten to insert the SD card and it's a waste of my time to power the system off again and restart hekate.

There should be three options (abort-retry-ignore style) if the SD card fails to mount:

- Cancel (return to main menu)
- Retry (try mounting the SD card again, now that the user has had a chance to insert it)
- Boot Stock (the current default)

fennectech, 2018-07-18:
I support this solution.

CTCaer, 2018-07-19:
Well cancel and retry fall on the same page.

So, something like, "Press PWR to Continue or VOL to go back to the menu" will do the job.

fennectech, 2018-07-19:
Unable to mount SDcard (is it inserted?)
Vol up to continue to stock
Vol down to retry

CTCaer, 2018-07-19:
@fennectech 
Added clutter. 
VOL and then PWR is not so much work for retrying booting.