CTCaer/hekate #50: File-level SAFE/SYSTEM/USER backup/restore

igniscitrinus, 2018-07-22:
Given the Switch's propensity to utilizing the microSD over the eMMC for storage purposes, it's common to find that the on-board storage is largely unused. 
Having the option to perform backups on a file level would significantly cut down the time required to perform a backup over block-level backups (Especially so for SAFE and USER).

CTCaer, 2018-07-22:
This was feature that a lot of thought went into it.

It requires the utilization of BIS keys and needs a greater code complexity.
Also a restore will utilize the formatting and directory traversing of FatFs, which will increase the binary size to great lengths.
This cannot fit, in hekate.

But a smart backup/restore option is nice to have.
So maybe this can be added as a payload **module** (now that payload launching was materialized internally).

So, I'll keep it open.

Serfrost, 2019-04-06:
I've been wondering about this for awhile and would like to say I think this would be greatly appreciated by anyone who keeps multiple backups, either for redundancy or being able to restore a rawnand.bin from previous firmware versions.

As it stands, the very first backup made on a fresh Nintendo Switch is indeed 29GB, but you can use an archive manager to ZIP the rawnand.bin with Normal compression and it will reduce the size of this backup to give or take 756MB.

When the sectors of the NAND have been used at any point in time, say, by installing a game, this makes compressing the rawnand.bin image impossible, at least up to the point of what was already written to; even if you System Format.

29GB backups add up, so I'd love to see an option to be able to make these smaller.

[image]

giorgi1324, 2023-06-09:
Here with the same feedback as @Serfrost

OLED full nand backups are pretty much uncompressable and take a lot of space.

Is there any chance we could get an intermediate solution for this where we can skip the USER partition from the backup? My very limited understanding is that there's nothing critical in it for the recovery of Switch.

CTCaer, 2023-06-09:
@giorgi1324 that's already a thing.
Just run the eMMC SYS option instead of GPP.

giorgi1324, 2023-06-10:
Oh amazing! I wish the guides out there mentioned this. Thanks for your fast response!

blvckmvgicdotexe, 2024-08-31:
@CTCaer sorry for resurrecting this issue, if I ever manage to break my system I'm not interested at all in the user partition (games etc.) so I was wondering if the system can be brought up to a working state with the system partition only (and maybe prodinfo, prodinfof etc.) aka with an eMMC SYS backup only; NxNandManager offers you to wipe USER in the final RAWNAND backup, does it make sense or it's a recipe for disaster?

impeeza, 2024-09-01:
@blvckmvgicdotexe this is not a support site, this is for Issues, bugs report please go to Discord or GBATemp.  even on GBATemp there are two tutorials to do what you are asking.

please close the isue.

blvckmvgicdotexe, 2024-09-01:
@impeeza I'm not asking about feasibility in terms of how to do it but rather if it is a good idea or not, in the current state of search engines it's hard to find an answer to this specific question
I'd rather have a working nx without saves or titles if I ever managed to break the system partition in some way than having a 32gb dead weight to upload on several cloud drives for backup

impeeza, 2024-09-01:
Ah, OK I understand now.

CTCaer, 2024-09-01:
`SYS` backs up all partitions, as partitions. Except USER.
The size of all these partitions is just ~2.5GB uncompressed.
USER only keeps game data and game savedata. All critical/important stuff are on the other partitions.
So yeah it can be used for that. You just restore the partitions when needed and wipe USER gpt partition with a tool via pc (that's in case you encounter a fatal because of mismatched stuff).

All in all, the most important partition is prodinfo that contains cal0. And also some calibration if Lite.
Anything else can be generated properly.