CTCaer/hekate #52: will you Bulid a slim version?

jcrorxp, 2018-07-24:
thankyou for you contribution,but I think the payload.bin size is more and more large.And I was not good to deposit into a MUC like trinket M0. So can we make  a branch with a Slim hekate CTCaer. just with base tools, like TX  payload.bin?

CTCaer, 2018-07-24:
The size of the payload can't go more than 123.5KiB, anyway.

Otherwise it breaks completely.

There is a small version in the works though.

mariogamer2, 2018-07-24:
I thought the max size was like 2ko? Why is it breaxking before?

tumGER, 2018-07-25:
The idea is to have a minimal version that has a really small size and fits onto everything @fennectech . It may sound stupid to some but there are devices that could potentially have problems fitting the whole hekate payload onto it's very limited storage (especially with the upcoming Signature Patches taking the payload to nearly max size).

I'd guess something like a payload that just chainloads the full hekate from SD could potentially be useful for small dongles/mod-chips.

fennectech, 2018-07-25:
i suggested that before but ctcaer shut that down rather quickly

mariogamer2, 2018-07-26:
You suggested it as for the general case, not for dongles and stuff.

Also, hekate will probably hit the max size soon. At that point some feature can be implemented in external payload ( including maybe dev stuff? unused?)

tumGER, 2018-07-26:
Yeah, there is no possible way to add new features once sigpatches are included due to the enormous size (well, there is still some room) so moving to a system with multiple payloads will be the only way to include other tools - Maybe something like a pure bootloader (without any tools to choose from) and a tool-payload? I don't think that we reached the point where new stuff doesn't have to be added anymore and even if, this size limit will always stay a problem that we will have to deal with. 

We can still take some precautions like removing "unnecessary" options like dumping certain parts such as user and sys, removing CTCaer's logo, removing the standard bootlogo, etc. but I don't think that this is the ideal solution.

People are willing to go the extra step and add stuff to there SD if that means that we would have more features and even if, most people use SDFiles packs that could just add them. 

Having one single payload may be cool but is it really worth the cost?

mariogamer2, 2018-07-27:
I don't even know if there is enough space to merge kip patching, with payload booting.

tumGER, 2018-07-28:
Correct, rajkosto did manage to get it under the limit ( [see here ](https://github.com/CTCaer/hekate/pull/51#issuecomment-407870378)) but we only have **27 Bytes** left - So making this repo modular is currently the only way to get other features in

rajkosto, 2018-07-28:
getting rid of the menu logo frees up around 10K that should be plenty for some additional features if required.

fennectech, 2018-07-28:
Could the logo be loaded from disk?

GGLinnk, 2018-07-29:
My suggestion :
- Use two different payloads :
 1. The first one for launching CFW only.
 2. And the second for dumping / restoring / print infos / etc.

CTCaer, 2018-07-29:
I believe that this discussion reached its end.
Mostly because it's a hit or miss suggestion thread and here is not the place for that. It's not a forum.
If you have an implementation though, be my guest and make a pull request.

Don't forget that what you are proposing, they already either worked internally or in the thoughts list **since the development of v3.0**. It is constantly hitting max size since then and simultaneously optimized. And I have several solutions since then. I don't upstream them because I don't want things to be a "let them and forget them".
This specific limit forced rajkosto to GREATLY optimize and future-proof (if someone wants to add many kips) his PR.

Anyway, imo the payload size limit is not something that concerns any non-contributor.
Even if someone has a modchip (there's a limit so..).
As I said, if you have a real suggestion on this matter backed with code, be my guest.
Just remember that many things are worked or already done internally for this (set it and forget it loader, plug-ins, etc).

CTCaer, 2018-08-06:
The answer is no.
If everything goes well, next version (v5.0) will also be a loader.
So you'll be able to load hekate in hekate.

CTCaer, 2018-08-17:
New version will be somewhat modular. This addresses this.

closed 2018-08-17 by CTCaer