CTCaer/hekate #53: Why keep "useless" functions ?

GGLinnk, 2018-07-29:
Hello,

I see that Hekate wants to add more and more functionality (that can be found elsewhere, like gpt restore or NAND restore, etc.) in a restricted size of payload instead of just only be what it's designed for (launching kip, patching them, etc.).

Also, why keep a copy of the splash screen inside of the payload instead of an external support (on SD Card) ?

So i ask, why keep functions that can be find elsewhere at the expense of main usage of hekate ?

Sorry if I look rough, I still have difficulties to express myself in English.

GGLinnk, 2018-07-29:
Or maybe use an external file to make a double sided payload, like SX do with boot.dat ?

CTCaer, 2018-07-29:
There's already a discussion about this in https://github.com/CTCaer/hekate/issues/52.

closed 2018-07-29 by CTCaer