CTCaer/hekate #54: More convenient joiner scripts

nkizz, 2018-07-30:
Using wildcards, you can use one script for 1GB parts and 2GB parts. You can also show progress with pv on unix, given it's installed.
join_linux_macos.sh
```bash
#!/usr/bin/env bash
if command -v pv >/dev/null; then
	cat rawnand.bin.* | pv -s 30g > rawnand.bin
else
	cat rawnand.bin.* > rawnand.bin
fi
```
join_windows.bat
```batch
copy /b rawnand.bin.* rawnand.bin
```

CTCaer, 2018-07-30:
Yeah these are better. The reason I had them hardcoded in the first place, was to make sure the order was always correct (fool proof way).

In *nix the default order is always by logical names though? Or it is by file chainlink like `ls`?
I want to avoid these types of reported corruptions on every *nix flavor.

nkizz, 2018-07-30:
According to the [bash man page](https://linux.die.net/man/1/bash):
 `After word splitting, unless the -f option has been set, bash scans each word for the characters *, ?, and [. If one of these characters appears, then the word is regarded as a pattern, and replaced with an alphabetically sorted list of file names matching the pattern.`
Since bash is specified, it will always be alphabetical.