CTCaer/hekate #55: add hardware tester

thomleg50, 2018-07-30:
it could be really cool for have a log with parts of the switch connected:

The gamecard module is connected to the motherboard

The sd module is connected

Screen is connected

etc

Thanks if you can

CTCaer, 2018-07-30:
But why?
Most of them can be seen right away if they work anyway.

thomleg50, 2018-07-30:
yes but for example my sd card don't work and idk if it's the module that is hs or disconnected

CTCaer, 2018-07-30:
If it works in Horizon, remove case and press the connector and try to dump BOOT0/1. If it works, it's just disconnected (some data pins).

Anyway this can't be identified, because the sd card semi works but the mode is not supported (1bit wide bus). If it doesn't work at all the check the connector again. It may be completely unsocketed.

closed 2018-07-30 by CTCaer