CTCaer/hekate #58: Push/pull file from SD a la adb

mariogamer2, 2018-08-04:
Remembered this not so long ago from a past issue.
Since we got module now, this would be good as one.

ValliantStorme, 2018-08-04:
It might be more convenient to just be able to launch memloader

mariogamer2, 2018-08-04:
Yeah maybe.
But I remember that CTCaer was kinda positive on it? I think it was on the original repo so it got deleted.

RenaKunisaki, 2018-08-06:
I wonder if there could be a sysmodule that implements adb push/pull?

CTCaer, 2018-08-06:
Sysmodule? As in HOS sysmodule? Wrong project.
If you mean a hekate module, then yeah it could.
But I will not develop it. 
(For me, my custom memloader's UMS is enough.)

fennectech, 2019-02-24:
does your custom memloaders ums launch sd automaticly?   Ive been looking for a memloader that doesnt load the menu

deurk, 2019-07-02:
> does your custom memloaders ums launch sd automaticly? Ive been looking for a memloader that doesnt load the menu

I'd be interested by it if you found something that worked.

CTCaer, 2020-05-01:
Added in version

closed 2020-05-01 by CTCaer