CTCaer/hekate #70: Please cancel KASLR on 5.0.0 or above.   Thank you very much!

Charlieeco95, 2018-08-14:
Please cancel KASLR on 5.0.0 or above.   Thank you very much!

tumGER, 2018-08-14:
That sounds really hard to do and would probably make the system unstable - Also what exact benefit would it bring?

Charlieeco95, 2018-08-15:
@tumGER  Thanks for replying my message.

KASLR is used for anti-hacking.
That's the reason why I wish it could be canceled.  or else, it's very difficult to find the real addresses or pointers to hack games.

Charlieeco95, 2018-08-15:
A reference  to whom it may concerned:

http://switchbrew.org/index.php?title=Memory_layout

kaslrBase = (rand64ViaSmc() % 0x3FFF0 << 21) + DRAM_VA(_start);

Can anyone please help to patched/disable this random value  (rand64viasmc)?   

Then, it would be all good.  Thank you so much in advance.

Thank you so much~

Charlieeco95, 2018-08-18:
@jakibaki 

Thanks so much for the explaination and your big help!

I will check it out.

AnalogMan151, 2018-08-23:
@jakibaki So basically  comment this part out? https://github.com/Atmosphere-NX/Atmosphere/blob/master/stratosphere/loader/source/ldr_nso.cpp#L185-L187

tumGER, 2018-08-25:
@AnalogMan151 
>which can be done in loader by keeping aslr_slide at zero 

It already stops doing that by default but you can re-enable aslr when changing that var:
https://github.com/Atmosphere-NX/Atmosphere/blob/44e2412ae6905924e9ed3fd228e3ceeda0b8d9d6/stratosphere/loader/source/ldr_nso.cpp#L184