CTCaer/hekate #71: Hekate payload success... Blank screen on switch.

colking, 2018-08-16:
Hey,
I'm not sure if its my jig, but I get a successful payload but no menu appears on the switch screen.  I'm on 5.1.0 and I tried a good quality 200gb scandisk sd card.  I tried exFAT and FAT32.  Can you suggest anything?  Thanks

CTCaer, 2018-08-16:
New switch?

It may be patched..

What's your serial (partial not all of it)
Is it one from these:
```
XAJ700438 to XAJ700439 
XAJ400541 to XAJ400564
XAJ700445
XAW100818 to XAW100857
XAW700183
```

colking, 2018-08-16:
Its between: XAW100818 to XAW100857

CTCaer, 2018-08-16:
Sorry, patched unit with non-exploitable RCM.
Even if you see that the file was sent correctly, the exploit fails.

You've done wrong to update it to 5.1.
You missed the warmboot exploit.

Better try to return it for whatever reason you can find.

closed 2018-08-16 by CTCaer

colking, 2018-08-16:
Damn! :(  I can return... Thanks for your help!