CTCaer/hekate #77: Hekate Softlocks itself

0xBootROMError8046, 2018-08-23:
When I enter the payloads/more configs menu without the corresponding files there's no back button and it  doesn't print an error, it just stays at a black screen so i need to force shutdown my switch.

CTCaer, 2018-08-23:
Will check on that. All this time I was so used with always having files there, that I've rarely checked these cases or checked by testers.
(I remember testing these though.. maybe I broke it with some specific directory list changes I did.)

A hotfix will not be provided though, because this release is mostly from pressure to not hoard payload/linux launching for longer, by having features added all the time..

There's another release in the works, that will move some parts of hekate to modules, ram training support, 64bit modules, a whole BDK (bootloader devkit) and many more..

For now, please always have some files there.

CTCaer, 2018-08-25:
https://github.com/CTCaer/hekate/issues/86

closed 2018-08-25 by CTCaer