CTCaer/hekate #78: Black Screen Loading SXOS Payload

darkunderdog, 2018-08-23:
I know this is going to get hate... I can load other payloads... but I get a black screen loading the SXOS Payload... Any Magic that needs to be done to load this payload?

It works fine if I boot directly from the payload and not through Hekate 4.0

CTCaer, 2018-08-23:
Well...

That's because of incomplete hw init and assuming stuff that sx loader does.

This will only be "fixed" because, hekate is intended as a bootloader that can launch anything.
If they change the loader in the future for various reasons, this will be dropped through.

Darky2k1, 2018-09-12:
Are the internal fixes planned for release soon? (Sorry Just Checking)

CTCaer, 2018-09-12:
yeah

CTCaer, 2018-09-19:
https://github.com/CTCaer/hekate/commit/89fb29e35e6bd0358066938d5e97c6b25958090f

closed 2018-09-19 by CTCaer