CTCaer/hekate #79: 4.0 issues

bakugo, 2018-08-23:
* If there are no boot entries in hekate_ipl.ini, the launch menu can't be used for payloads or extra inis either
* Selecting "payloads" or "more configs" when there are no items in the corresponding SD folder results in an empty screen that can't be exited
* Payload launcher doesn't work with some payloads, it just turns off the screen and the console must be force shut down (happens with fusee example and SX loader)

Was this release tested at all?

CTCaer, 2018-08-23:
[image]

https://github.com/CTCaer/hekate/commit/5328f89ffe356c5c89e21e39adbfcf2c198c6095
You had 2 full days and a half to test it! What did you do all that time??
You are not paid to lay around and sleep..

CTCaer, 2018-08-25:
https://github.com/CTCaer/hekate/issues/86
https://github.com/CTCaer/hekate/issues/78
https://github.com/CTCaer/hekate/issues/77

Also about the payloads, fusee is currently broken anyway.
Hekate can boot exosphere like before.

closed 2018-08-25 by CTCaer