CTCaer/hekate #81: Hide "launch payloads" and "more configs" entries and have a config param to launch an ini config

jlozes, 2018-08-23:
First, thank you for all the work you do for this payload, the V4 is realy impressive.

Like I say in the title of this issue, I want to propose some new features.

The first one is the possibility to hide the two menu for launching payloads and for accessing others configurations of the "/bootloader/ini" dirrectory. For the payloads launching menu it should be easy to do and we have the new param "payload=path_to_payload" to launch the one we want.

The second thing is to have a param on the config file to load a specific ini configuration (an ini like those witch are in the "bootloader/ini" folder, without the global config values).

I will explain what I suggest by an example of config file, based on SDFilesSwitch of TumGer:

"
[config]
autoboot=0
bootwait=5
customlogo=1
verification=2
show_launch_payload=0 or 1
show_more_configs=0 or 1
{-- Custom Firmwares --}
[ReiNX]
payload=bootloader/payloads/ReiNX.bin
[Atmosphere via Hekate]
configs_file=bootloader/ini/atmosphere_hekate.ini
[CFW via Hekate]
kip1=modules/newfirm/loader.kip
kip1=modules/newfirm/sm.kip
kip1=modules/newfirm/pm.kip
kip1patch=nosigchk
[CFW + Prevent GC access]
kip1=modules/newfirm/loader.kip
kip1=modules/newfirm/sm.kip
kip1=modules/newfirm/pm.kip
kip1patch=nogc
kip1patch=nosigchk
{ }
...
"

I ask if this can be done because I can't read the screen (I'm quasi blind) and it is difficult to learn the "launch payloads" and "more configs" menus because they use files name to display options order.

Again, thanks for your hard work.

CTCaer, 2018-08-24:
I understand this. It's like an #include.

We'll see about these 3.

jlozes, 2018-08-24:
Yes, something like that, at least to hide or show the two menus.

For the param "configs_file" that I propose I think that the config must not appear directly in the main menu but when you click on the entry, the config menu targeted is loaded like a secondary config file. If you prefer it's like a folder containing one set of configs. The only problem that come in my mind with this is that you probably should enable this only for the main config file because it's not useful to have "subfolders" witch could be more problematic than useful.

CTCaer, 2018-08-24:
Oh. Not as include.. but like a lua. So .ini depicts the menu.
To have menus created by ini, is bad design for hekate.

It will mess with autoboot settings and config save functions.

On the other hand, grouping is a nice feature to have. (well I only have 4 main entries, so it's not important for me).

jlozes, 2018-08-24:
I haven't think of autoboot, it's a problem to apply my idea effectively.

CTCaer, 2018-08-25:
As long as accessibility is concerned I don't know how hiding these options will help.
(The configs are always 2 VOL DOWN away)

In my opinion, it's better to change your hekate_ipl.ini in a PC to have your most used entry at the bottom.
This way you can go by doing this sequence when in main menu: 

1. **PWR** (Enter Launch menu)
2. **VOL UP** (goes to last entry)
3. **PWR** (launches that configuration)

I'll mark this issue as informative for now and see if we can come up with a better solution.

jlozes, 2018-09-02:
The problem is that in the configs list, if you press "Vol up" when you are on the first entry of the list, the selection doesn't go to the last config of the menu but stay on the first entry. After it's not realy a problem, I even think that it's good because like this, if you are lost on the list, you can press "Vol up" multiple times ans you'll know that you are to the begining of the list.

With the new Hekate you only have to entries to pass with "Vol down", it's not so difficult after all to press three times "Vol down" to select the first entry of the main config menu.

CTCaer, 2018-09-03:
It should.
I'll check it.

CTCaer, 2018-09-08:
> The problem is that in the configs list, if you press "Vol up" when you are on the first entry of the list, the selection doesn't go to the last config of the menu but stay on the first entry.

Just checked and this does not happen. It goes to the last entry. Do you have a specific setup that I can test?

jlozes, 2018-09-12:
Here is my main ini:

"
[config]
autoboot=0
bootwait=5
customlogo=1
verification=2
{-- Custom Firmwares --}
[CFW]
kip1=modules/newfirm/loader.kip
kip1=modules/newfirm/sm.kip
kip1=modules/newfirm/pm.kip
kip1patch=nosigchk
[CFW + Prevent GC access]
kip1=modules/newfirm/loader.kip
kip1=modules/newfirm/sm.kip
kip1=modules/newfirm/pm.kip
kip1patch=nogc
kip1patch=nosigchk
{ }
# I wanted to call this tomNX for fun but that would have destroyed guide compatibility :P

{----- FTP-Server -----}
{Port: 5000}
{Dont run SDFilesUpdater with this!}
[FTP + CFW]
kip1=modules/newfirm/loader.kip
kip1=modules/newfirm/sm.kip
kip1=modules/newfirm/pm.kip
kip1=modules/sys-ftpd.kip
kip1patch=nosigchk
[FTP + CFW + Prevent GC access]
kip1=modules/newfirm/loader.kip
kip1=modules/newfirm/sm.kip
kip1=modules/newfirm/pm.kip
kip1=modules/sys-ftpd.kip
kip1patch=nogc
kip1patch=nosigchk
[FTP + CFW + Newest LayeredFS]
kip1=modules/newfirm/loader.kip
kip1=modules/newfirm/sm.kip
kip1=modules/newfirm/fs_mitm.kip
kip1=modules/newfirm/pm.kip
kip1=modules/sys-ftpd.kip
atmosphere=1
kip1patch=nosigchk
[FTP + CFW + Newest LayeredFS + Prevent GC access]
kip1=modules/newfirm/loader.kip
kip1=modules/newfirm/sm.kip
kip1=modules/newfirm/fs_mitm.kip
kip1=modules/newfirm/pm.kip
kip1=modules/sys-ftpd.kip
atmosphere=1
kip1patch=nogc
kip1patch=nosigchk
[FTP + CFW + Older LayeredFS]
kip1=modules/newfirm/loader.kip
kip1=modules/newfirm/sm.kip
kip1=modules/oldlayered/fs_mitm.kip
kip1=modules/newfirm/pm.kip
kip1=modules/sys-ftpd.kip
atmosphere=1
kip1patch=nosigchk
[FTP + CFW + Older LayeredFS + Prevent GC access]
kip1=modules/newfirm/loader.kip
kip1=modules/newfirm/sm.kip
kip1=modules/oldlayered/fs_mitm.kip
kip1=modules/newfirm/pm.kip
kip1=modules/sys-ftpd.kip
atmosphere=1
kip1patch=nogc
kip1patch=nosigchk
{ }

{------ LayeredFS -----}
[CFW + Newest LayeredFS]
kip1=modules/newfirm/loader.kip
kip1=modules/newfirm/sm.kip
kip1=modules/newfirm/fs_mitm.kip
kip1=modules/newfirm/pm.kip
atmosphere=1
kip1patch=nosigchk
[CFW + Newest LayeredFS + Prevent GC access]
kip1=modules/newfirm/loader.kip
kip1=modules/newfirm/sm.kip
kip1=modules/newfirm/fs_mitm.kip
kip1=modules/newfirm/pm.kip
atmosphere=1
kip1patch=nogc
kip1patch=nosigchk
[CFW + Older LayeredFS]
kip1=modules/newfirm/loader.kip
kip1=modules/newfirm/sm.kip
kip1=modules/oldlayered/fs_mitm.kip
kip1=modules/newfirm/pm.kip
atmosphere=1
kip1patch=nosigchk
[CFW + Older LayeredFS + Prevent GC access]
kip1=modules/newfirm/loader.kip
kip1=modules/newfirm/sm.kip
kip1=modules/oldlayered/fs_mitm.kip
kip1=modules/newfirm/pm.kip
atmosphere=1
kip1patch=nogc
kip1patch=nosigchk
{ }

{-------- Stock -------}
[Stock]
[Stock + Signature Patches]
kip1patch=nosigchk
[Stock + Prevent GC access]
kip1patch=nogc
[Stock + Signature Patches + Prevent GC access]
kip1patch=nogc
kip1patch=nosigchk
{ }

{---- Experimental ----}
[Atmosphere]
secmon=modules/atmosphere/exosphere.bin
kip1=modules/atmosphere/loader.kip
kip1=modules/atmosphere/sm.kip
kip1=modules/atmosphere/fs_mitm.kip
kip1=modules/atmosphere/pm.kip
logopath=modules/atmosphere/atmosphere.bmp
atmosphere=1
[Atmosphere + Signature Patches]
secmon=modules/atmosphere/exosphere.bin
kip1=modules/atmosphere/loader.kip
kip1=modules/atmosphere/sm.kip
kip1=modules/atmosphere/fs_mitm.kip
kip1=modules/atmosphere/pm.kip
logopath=modules/atmosphere/atmosphere.bmp
kip1patch=nosigchk
atmosphere=1
[Atmosphere + Prevent GC access]
secmon=modules/atmosphere/exosphere.bin
kip1=modules/atmosphere/loader.kip
kip1=modules/atmosphere/sm.kip
kip1=modules/atmosphere/fs_mitm.kip
kip1=modules/atmosphere/pm.kip
logopath=modules/atmosphere/atmosphere.bmp
kip1patch=nogc
atmosphere=1
[Atmosphere + Signature Patches + Prevent GC access]
secmon=modules/atmosphere/exosphere.bin
kip1=modules/atmosphere/loader.kip
kip1=modules/atmosphere/sm.kip
kip1=modules/atmosphere/fs_mitm.kip
kip1=modules/atmosphere/pm.kip
logopath=modules/atmosphere/atmosphere.bmp
kip1patch=nogc
kip1patch=nosigchk
atmosphere=1
{ }

{----------------------}
{ }
{Running v8.0.1 of tumGER/SDFilesSwitch}
{By: tomGER(@tumGER)}
{StevenMattera, WerWolv}
{ }
{------ Thanks to -----}
{ }
{Reswitched Team, CTCaer}
{Thog, vgmoose, Brawl345}
{GGLinnk, jpe230, Adubbz}
{thomasnet, TheDgtl, Rei}
{jakibaki, rajkosto,}
{mariogamer2 and many more}
{ }
{----------------------}
{ }
{Press R + Album to launch Album instead}
{Use sigpatch_payload.bin for Signature Patches}
{ }
{Discord: https://discord.gg/qbRAuy7}

# This code isn't political -- It isn't even code

"

CTCaer, 2018-09-12:
put it in code quotes: ```

CTCaer, 2018-09-19:
Clean that ini. Too much unneeded stuff.
It probably also breaks the max boot entries limit.

Don't forget to try the new v4.1. I fixed many bugs with the ini listing.

closed 2018-09-19 by CTCaer

jlozes, 2018-09-19:
I know that it as to many entry but it's also the purpose of tests, to sea what could be buggy.

You've closed my issue, do you think that other things that I propose will not be worked on (for hiding "payloads" and "more configs" options)?

CTCaer, 2018-09-20:
Yeah I forgot to tell you about it.

Because the last entry is a caption, going to last entry does not work.
This is a limitation that intent to keep (because changing it needs to change the whole entry selection logic).

Also about the hide the payloads/more configs, I don't intend to add them.

My statement that this ini needs a clean up stands true though.
Entry and comments exceed the 50 char limit and it has too much unneeded stuff.
Anyway, a simple fix is to add a new [boot entry] after everything.

jlozes, 2018-09-20:
OK, thanks for the informations, it's realy instructive.

For the other functions that I've proposed I understand and to be honest, it's not realy a problem because now I know witch entries are always displayed, counting is not so dificult after all.

Thanks for all.

CTCaer, 2018-09-24:
@jlozes 
just to inform you, I internally fixed the go back to start/end when a caption/change line is used as first or last entries.

jlozes, 2018-09-25:
Thanks for the info and for your work... again.