CTCaer/hekate #83: Add possibility to use joker "*" in kip1 param path for file name

jlozes, 2018-08-23:
It could be intresting to have this possibility to simplify the configuration file.

For example:

"
[all modules]
kip1=modules/newfirm/loader.kip
kip1=modules/newfirm/sm.kip
kip1=modules/newfirm/pm.kip
kip1=modules/sys-ftpd.kip
kip1=modules/xor.play.kip
kip1=modules/sys-netcheat.kip
kip1=modules/switch-rich-presence.kip
kip1patch=nosigchk
"

This one could be:

"
[all modules]
kip1=modules/newfirm/*.kip
kip1=modules/*.kip
kip1patch=nosigchk
"

Thanks.

CTCaer, 2018-08-24:
hmmm not a bad idea.

But it's needed? I mean that it's a "set it and forget it" thing the ini configuration.

jlozes, 2018-08-24:
Needed no but it could be useful in some case, for example to test more quickly a set of modules.

CTCaer, 2018-09-19:
My current plan btw, is for this to be worked on for a later version.

CTCaer, 2018-09-24:
https://github.com/CTCaer/hekate/commit/1392e6eaf49b4f593e5b3ee42c76993c45f638d8

closed 2018-09-24 by CTCaer

jlozes, 2018-09-25:
Thanks, this will be useful.