CTCaer/hekate #91: Kezplez

1Canna, 2018-08-27:
Hi there, Ctcaer 

I have updated my files in my backup guide for users on the gba temp community,
With the latest hekate 4.0 and tomger sd files, And when the users following my guide try to launch kezplez  it will not run and  it will error on launching... 

Kezplez works fine with hekate 3.2 and associated files, can you tell me the solution or what has changed in 4.0 for it not to work thanks

CTCaer, 2018-08-27:
Kezplez is in the HOS side, so hekate does not touch it's functioning.

Have in mind that sdfiles have a newer atmo revision.

I can help if you give me the full test cases (3.2 vs 4.0).

1Canna, 2018-08-27:
I have found the solution,, It seems that your new version outputs tseckeys.txt
Were as the old version outputs tseckey.txt
Would you be so kind as to change this for us pretty please 

I also understand that the dump.pck1,pckg2.restore,boot1,boot2 will have to be moved into backup folder but thats no problem .. but i would like to request you change tsekeys.txt back to tseckey.txt

1Canna, 2018-08-27:
So basically,
If i rename tseckeys.txt to tseckey.txt  
And move Dumps, Pkg1, pkg2, restore, boot1, boot0 to backup folder root
The kezplez app will work correctly.

Issue is due to Nand Id folder         
Folder structure  
Backup>nandIdfolder>dumps pckgs etc        Notworking
Backup>Dumps pckgs etc                               Working

And tseckeys.txt needs change of output name to read tseckey.txt

Thank you 
What can you change ? as i will update my guide For the members

CTCaer, 2018-08-27:
I can easily change tsec_keys.bin to tsec_key.bin but for serial number I can't do it because it's a nice feature for identifying the backup.
I believe the best way is to have a heuristic that enters or asks which folder you want to process in backup folder.

1Canna, 2018-08-27:
Yes would you be so kind as to rename the tsec .bin for me thank you, and update me when you release that version, as for the folder for now i will tell the members to simply move the contents to backup thanks

CTCaer, 2018-08-27:
You can ping **tesnos** if you want.
Usage of **fsdev_diropen** will meet the demands of v4.0.

1Canna, 2018-08-31:
Hi there CTCaer hope your well....
Did you update the hekate4.0 to change the filename output of tsec_key yet ?

CTCaer, 2018-09-01:
https://github.com/CTCaer/hekate/commits/master

Nope not yet.

Fallenleader, 2018-09-02:
To clarify what 1Canna was saying, Hekate dumps the files into a folder with some letters and numbers (e.g. a6b78c9).
For those having issues:
rename tseckeys.txt to tseckey.txt
Copy all files and folders inside your weirdly named folder and paste it in Dumps.
here is a picture of what your final result should look like for clarity, as 1Canna had me confused at first on what I was supposed to move to where.
[image]

1Canna, 2018-09-02:
@shchmue 
Thanks again gorgeous closing issue as solved,

closed 2018-09-02 by 1Canna

CTCaer, 2018-09-03:
@shchmue work is what I suggested.
Gj