CTCaer/hekate #92: Linux Chainloading for multiple versions of linux.

GavinDarkglider, 2018-08-27:
I am trying to port a different version of linux other than Lakka. From what I can tell from Lakka, it loads the coreboot.rom directly, which is awesome. Good Work there. My question is actually a 2 part question. How does Coreboot/U-Boot handle the boot.scr? On painless-linux, the boot.scr was in the boot directory, in lakka switch, it is in the lakka/boot directory. If I try to do this with my gentoo build, which boots fine using shofel, I get nothing. Will try putting my custom boot.scr in the lakka/boot directory later.

This leads me to my next question.... Is there anyway, that we can get hekate to dynamically create the boot.scr, and just put the kernel, dtb, and args directly into the configuration.ini. It would be easier to edit an INI file, then it is to generate and copy a new scr file everytime I want to test a different kernel paramater. Eventually, I also want to see about porting the hardware patches to an android kernel, to see if the console will run that, and a feature like this would be nice for development.

At the very least, if this requires modifications to U-Boot, or coreboot, or any information you can give me on this topic, would be helpful.

CTCaer, 2018-08-27:
u-boot is responsible for partition or image loading + using a command via boot.scr.

Boot.scr is a compiled (?) binary and hekate can't do this.

For android, you need to use coreboot that will mount your partitions or a disk image with partitions. By adding the needed android kernel functions, you can **eventually** have android in switch.

GavinDarkglider, 2018-08-27:
I get this much, but that doesnt help with the fact that I still dont know exactly where uboot is looking for this boot.scr, or how to change where, so I can have multiple linux installs, or android, on one sd card.

Drakulix, 2018-08-27:
> I get this much, but that doesnt help with the fact that I still dont know exactly where uboot is looking for this boot.scr, or how to change where, so I can have multiple linux installs, or android, on one sd card.

This is the relevant commit changing the default behavior of u-boot for lakka:
https://github.com/lakka-switch/u-boot/commit/9f08090e970178c77f148e2440f5e55b94690663

Instead of running the usb server, like failoverfl0w's port it is running a build-in command:
https://github.com/antmicro/u-boot/blob/master/include/config_distro_bootcmd.h

Which basically scans partitions until it finds a boot.scr. It will always boot the first one found, so what you would have to do is probably changing the default command, so it will always boot your boot.scr and use that u-boot build for your second linux payload.

GavinDarkglider, 2018-08-28:
Thank You, I can work with that, just means I need a seperate coreboot payload for each distro, which is sleightly annoying,  with the boot_prefixes path changed to where ever I decide to keep the boot.scr and other files for my distro.