CTCaer/hekate #98: [Feature Request]Add Hotkeys for Autoboot Shortcuts

annson24, 2018-09-01:
Well, payload chainloading is now possible, maybe multiple hotkeys is now a good idea?

CTCaer, 2018-09-01:
I thought of this before, but it cannot be done without creating problems.
Many users go into RCM using PWR + VOL+

This can be caught by the button parser. That's why the go into menu is VOL-

Maybe a touch gesture can be added in the future..

annson24, 2018-09-01:
So Vol+ and Vol- are the only keys we can use for this (And PWR?)? I always had the impression we can use any keys. Turns out I was wrong? Is that why up until now hekate still only uses those three keys to navigate in the menu?

GavinDarkglider, 2018-09-01:
To answer your question simply, it is easier to use the GPIOs from the CPU that go to volume, and power buttons then it is to write a driver, and compress it into the size constraints of the initial payload. You could probably write the driver as an auto-loadable module once hekate boots, but that kind of defeats the purpose of quick launch, if you have to wait for the boot loader to load a driver every time the console boots. Personally, I am waiting for the touch screen module, then I will probably never use auto boot. I like the idea of choosing what I want to boot, since there are multiple firmwares to play with, and linux.

CTCaer, 2018-09-03:
We can also use HOME (jig). But as Gavin said, we can't just write a whole joycon driver only to have button combos. (Although with the GUI it will make sense).

closed 2018-09-03 by CTCaer