CTCaer/hekate #99: [Enhancement] Key combo to terminate nand dumping/check for existing before starting

Fallenleader, 2018-09-01:
After a long nand dump, I thought I had dumped the wrong one, and triggered a second one. Not only did I lack space, it started dumping a partial without confirmation. This was the difference between me taking the switch with me earlier, and having to leave it at home on charge because I didn't have the time for another full nand dump (granted had I needed to dump it, I wouldn't have minded as much).

This could have been resolved by:
checking if a nand dump already existed (bonus: having some sort of file to check if the dump was valid, like a MD5 or something, so the user could delete and redump, or at least be made aware the dump was bad)

Prompting confirmation to dump from user (bonus: Prompt before overwriting a nand dump, and if the user agrees, simply overwrite the existing dump)

Prompting user to confirm for a partial dump, which in general is not a good idea to automatically do in the first place (people are stupid, and may not be aware the dump is a partial).

Providing a termination key combo if the user initiates an unwanted dump.

1Canna, 2018-09-02:
You could of avoided by having enough space, not cancelling a dump cause its a bad idea,
and deleting the dump you cancel, just because you made mistakes does not mean you should ask the maker to change what works when used correctly

CTCaer, 2018-09-03:
Well from all these the most requested and one that makes sense is the backup cancellation.
Happened to me when testing, where I forgot to disable verification and I know the frustration.

I'll think of a way (probably the btn timeout function with 0ms delay). I hope it will not raise backup time.

Fallenleader, 2018-09-03:
This was purely QoL and convince suggestions that would be a good idea to have in general.

I would gladly trade 10 minutes for an option to cancel in the event I select the wrong option to dump.

@1Canna I never said it was his fault, and these suggestions make absolute sense for a long term use of the application.

> not cancelling a dump cause its a bad idea,
Evidently, you didn't read my post correctly. Terminating a dump is not possible at this time, which was the point of this suggestion for Hekate.

> just because you made mistakes does not mean you should ask the maker to change what works when used correctly
No, but from a QoL standpoint, it does. These were similar if not identical things seen in godmode9 for the 3DS, so it isn't far-fetched to eventually get a request for this. I just happened to be the idiot to give him a use case reason for these enhancements :P

1Canna, 2018-09-03:
@Fallenleader 
You need to chill out....
I never said that you said it was his fault, you evidently did not read what i said correctly,
and i said cancelling a dump because you started another dump, and mentioned that there should be a feature to delete the old, were as that is a good idea,but you had no space left and you yourself made the mistake of not deleting the old...
I have no beef, and i think its a great idea being able to cancel a dump with a button combo or confirmation... Or maybe even do you wish to overwrite... 
Thank you for the response and good idea end of conversation.

Fallenleader, 2018-09-04:
> You need to chill out....

Wow. You have come off disrespectful TWICE now. My intentions in my reply were not such, and if they came across that way, I do apologize. That said:

> Thank you for the response and good idea end of conversation.

What gives you the right to jump in here an make a comment like you own this project? Heck even the initial response from you was borderline inserting words in CTCaer's mouth. Unless I missed the memo, this is his ship, his puppy, If he has an issue with my request, he has every right to dismiss it (which apparently he liked it). I was pointing out a use case where the feature requests I was asking would be a legitimate circumstance. I don't believe I asked for anything unreasonable, nor did I blame him for my mistakes. I simply mentioned that this situation happened, and having these features could prevent it in the future, as I am sure that if I can make that mistake having been modding my consoles since modchips and disc swaps were a thing (PSX, PS2, Wii, WiiU, PSP, PS3, 3DS, and currently my Switch, along with 11 years in IT), anyone else can to, especially since things are getting more and more involved to do each generation, and people are less and less inclined to learn things properly.

If you do have beef with me, feel free to PM or even email me, rather than having a ping pong about who is right, wrong, mad, or not on someone else's git. It would do better to reserve this space for the ticket and related information on it once he gets around to either implementing it, or dropping it one.

CTCaer, 2018-09-19:
https://github.com/CTCaer/hekate/commit/efc3fc34de096817eee3d388585f9fbaa3a227dc

closed 2018-09-19 by CTCaer