CTCaer/hekate #102: Partial NAND backup verification failure on rawnand.bin.14

JoshRickers, 2018-09-04:
Partial NAND backup using a 16GB SD Card. Verified as working with h2testw,Test finished without errors.
First batch completed successfully and asked me to clear some space to resume.
Moved the files across to computer and left the idx file in place.
Second batch completed successfully and asked me to clear some space to resume.
Moved the files across to computer and left the idx file in place.
Finally rawnand.bin.14 was dumping but it would fail after 4 - 5 minutes with a verification error.
I tried this multiple times, always producing the same error.

I then attempted it all again after a reboot. Starting from scratch. Got to the rawnand.bin.14 and it started with the same error.

I did notice that the filesize for rawnand.bin.14 was 2GB.

I edited the idx file to start off at rawnand.bin.13.

rawnand.bin.13 and rawnand.bin.14 created and verified successfully with rawnand.bin.14 being roughly 1.7GB in file.

I then ran the joiner script which created the rawnand.bin at 29.1GB (31268536320 bytes)

It looks like if you only create rawnand.bin.14 it creates a 2GB file where it should create the 1.7GB which is why the verification fails?

CTCaer, 2018-09-04:
Ok I know what's wrong.

The last one must be **1.12GiB**.
It seems that I didn't add a special case, when the partial backup is resumed from the last file. And because of that, it tries to create the wrong file size.
That bug was introduced with the pre-allocation that greatly speeds up writes to SD, especially on FAT32.

Until I fix it, this **can be mitigated on 16GB cards**, by adding a ~2GB dummy file. So the last part will fall on the part size decider.

CTCaer, 2018-09-19:
https://github.com/CTCaer/hekate/commit/efc3fc34de096817eee3d388585f9fbaa3a227dc#diff-2f6de2c63616e79a4a7009a6ae7d7eb8R1091

closed 2018-09-19 by CTCaer