CTCaer/hekate #103: Add warning text when there's problem with sd reader

balika011, 2018-09-05:
If the sdcard slot not seated properly there might be an issue that the sd works under hos, but not in hekate, because 1bit mode is not supported. In this case we can show a "nice" warning to the user to check the connection.

CTCaer, 2018-09-05:
As discussed, I may add support for 1bit in the end. But with a huge warning text.
It's true that the current sd init failed msg is confusing and does not help at all.

I'll also revise the error context in sdmmc. So it will return different error code.

CTCaer, 2018-09-09:
In the end this is implementation specific.
Can't be used for all sd cards.

I just changed the warning to inform for a possible sd reader problem.

CTCaer, 2018-09-19:
lel
https://github.com/CTCaer/hekate/commit/780736591e051a216914c8d0fd1c82d0a502fbc9#diff-2f6de2c63616e79a4a7009a6ae7d7eb8R98

closed 2018-09-19 by CTCaer