CTCaer/hekate #106: Can't connect to PC anymore

bnxflazy, 2018-09-08:
I did what you said, restore the BOOT0/1 with the firmware image, right now it's dead, can't boot it up , can't connect to internet, what else can I do ?

CTCaer, 2018-09-08:
12s power button..

bnxflazy, 2018-09-09:
What do you mean

CTCaer, 2018-09-09:
You said can't boot it up. You mean to boot HOS?

bnxflazy, 2018-09-09:
No, I mean can’t inject payload, can’t connect to PC, but after 12s press on power button, it works right now, but sadly I restore the BOOT0/1 which is from someone else , right now it shows
Initializing…
Identified package1 (“20180220163747).
Keylob version 4
Loaded package1 and keylob
And it stucks at here. I think it’s because the BOOT0/1 are not correct, right?  And right now I lost the my BOOT0/1, can I still recover it ?

bnxflazy, 2018-09-09:
> try to run it as stock.
>
> Stock is basically a boot entry in ini that has nothing.
>
> [Stock]
> #Nothing here

When I run it as stock, it shows the same info as I run CFW

CTCaer, 2018-09-09:
First of all, check your sd card is ok.
Use h2testw. (Backup your sd before that).

bnxflazy, 2018-09-09:
Yes, the sd card has no problem, and I followed the exact process on https://gbatemp.net/threads/how-to-install-run-any-switch-firmware-unofficially-without-burning-any-fuses.507461/, but still it doesn’t work, just showing initializing  packaged identified bla bla , what can I do now.

CTCaer, 2018-09-09:
You probably damaged your keyblobs.
You need to restore these. They are at x180000 of boot0.

bnxflazy, 2018-09-09:
Really? How to restore it....

bnxflazy, 2018-09-09:
I actually followed that guide, and deleted all my system file, and copied the 5.10 firmware to my system, why would the keyblobs broke…

CTCaer, 2018-09-09:
From a previous restore of a corrupted BOOT0

bnxflazy, 2018-09-09:
So you mean I broke the keyblobs when the first time I restore the corrupted BOOT0/1? But why I didn’t fix it when I try to restore everything according to that page.. so right now how could I do with it. Please ,this bothers me two days, can’t even sleep….  Thanks for replying me again.

CTCaer, 2018-09-09:
This cannot be fixed with the guide.
You need to restore them from a working boot0, which ofc we can't share, even if it's not unique.

bnxflazy, 2018-09-09:
You said They are at x180000 of boot0. Do you mean that the keyblobs are inside the boot0 file? Only boot0? And it’s not unique, it means that I can acquire it from others’ working boot0, right? But if I can get this, how could I restore them? You said the keyblobs from boot0 cant be shared even if it’s not uniqur? So how could I restore them if I do not have my own boot0, does this mean I can’t fix my switch anymore…..so sad, or you mean that you can’t share, but I can still get it from others’ boot0 ?

CTCaer, 2018-09-09:
You can still get them from other's boot0.

bnxflazy, 2018-09-09:
Thanks for the information, but what exactly is keyblobs, how can I get it from the boot0….. you said it’s at x180000, I have no idea what it that, boot0 is just a file, x180000, is this an address? Then keyblobs should just be few lines of code, right? I don’t know, really sorry for so many questions and bothered you so long……

bnxflazy, 2018-09-09:
How could this happen when I restore the boot0/1 using choidujour, shouldn’t it restore the whole boot0/1 file? Can my switch be fixed?

CTCaer, 2018-09-09:
This is an address. The boot0 that is restored from choi is 1.5MB (full BOOT0 is 4MB).

http://switchbrew.org/index.php/Flash_Filesystem#Boot_Partitions

bnxflazy, 2018-09-09:
Thanks for the information, but I just found on the page, it says that “Each one is unique compared to the others and they are all console unique. This is officially known as the "EKS" (encryption key source) area.” . Can I restore it from someone else?

http://switchbrew.org/index.php/Flash_Filesystem#Boot_Partitions

bnxflazy, 2018-09-09:
Okay, really thanks for your help, I’m not hoping that you could help me to solve this, But I do have last question, can my switch be fixed? If it can, then I will just try my best to read everything I need, and recover it by myself. If not, then I may give it up. Right now the situation is that I can not boot the original Nintendo firmware, and when I try to use hekate to boot it , it failed to launch the firmware and pop up that message. These are all I have faced, and I just want a confirmation, can I still recover it and launch the firmware or not. Thanks very much

http://switchbrew.org/index.php/Flash_Filesystem#Boot_Partitions

CTCaer, 2018-09-09:
You can reconstruct the current known ones (0-5) if you have your keys.
There's still hope. Except if you corrupted your PRODINFO (cal0). Then this is the end.

Unfortunately I don't have time to help you more or guide you. You'll find more helpful hands in forums.

bnxflazy, 2018-09-09:
You really scared me…. How  do I check whether or not the PRODINFO is corrupted. Will it be corrupted when I did the choi things two times?

CTCaer, 2018-09-09:
Well, if you used choice, you could access the rawnand partitions through the BIS keys.
So you are probably fine and only corrupted BOOT0/1

bnxflazy, 2018-09-09:
Thanks very much about it !!!

bnxflazy, 2018-09-10:
Could you briefly explain how can I extract the keyblobs from the boot0? Figured it all night, still can’t reach it

closed 2018-09-10 by bnxflazy