CTCaer/hekate #108: restored a bad BOOT0

bnxflazy, 2018-09-10:
Okat, here is the issue I found, I use Online TrID File Identifier to identify my own boot0, which is the one I restored and cause my switch "bricked", and I soon found that the boot0 file only contain two part, one is XEX and it's 66.67% . another is MP3, and it weight 33.33%.  but when I try to identify the others' normal boot0, it's totally different from mine. 
So I would like to explain the process agian, first of all, my switch worked fine, but I restored that bad BOOT0. then caused it "bricked", and it prompt "could not identified package1" as you knew, because that boot0 only contain two parts which is not correct.
And then I restored a boot0 from someone else, and it didn't help for sure, but it then prompt "package1 loaded" something
the third part I did was, I restored the bad BOOT0 back again. and then used the choi to restore the BOOT0 you told me. but after the process was done, it still prompt that "package1 loaded" thing, and still can not be booted.

So finally, can I still recover it ? cause right now the issue is much more clear, the backed up boot0 isn't correct at the first time, not even close, I just restore some trash I think. please help

fennectech, 2018-09-10:
I believe boot0 is not console specific. If it is.   Your boned.   You now have a lakka only switch.   If you can convince it to perform a firmware update you may be able to save it.      Try taking an boot0 from another users backup.  I’d send you mine but its in another city.

bnxflazy, 2018-09-10:
It doesn't help if I try to restore the boot0 from other's. I've already tried as I mentioned....

fennectech, 2018-09-10:
Other than that.  Try to boot with hekate as it replaces package1.    If you can manage a firmware update you might be able to recover.  6.0.0 is to hit soon

bnxflazy, 2018-09-10:
I think you have no idea what I'm facing... ofc I'm booting it with hekate, but when I launch the firmware, it stuck at "package1 loaded  package and keybolb"

fennectech, 2018-09-10:
With all of this bricking I’m going to take a fresh backup xD

fennectech, 2018-09-10:
Actually.  Read this.  There may be hope yet if prod info isn’t fucked 
https://github.com/CTCaer/hekate/issues/106#event-1835459860

bnxflazy, 2018-09-10:
that's actually what I issued

fennectech, 2018-09-10:
I recommend joining the discord.  Unbricking isnt my strong point.  And this Earle its not a sure fire thing.     But. Without the keys.   There is nothing left but to put lakka on it

bnxflazy, 2018-09-10:
What is discord and Earle,  and what do you mean without the keys?

fennectech, 2018-09-10:
Recovering your switch isnt a sure thing.   Discord is a chat service.

bnxflazy, 2018-09-10:
I won't give it up, there must be a way to fix it

fennectech, 2018-09-10:
Without your keys there isnt.

bnxflazy, 2018-09-10:
What's the key?

fennectech, 2018-09-10:
The encryption keys in your switch.

fennectech, 2018-09-10:
You can still use lakka

bnxflazy, 2018-09-10:
Do you mean the BIS key? if yes, I have it actually, there is no sense to use that, and I want to recover my switch not only I wanna use it , but I also enjoy the process

fennectech, 2018-09-10:
Well. Your going to have to figure out how to build a new boot0 image with your keys.  That’s way beyond me and out of the scope for this forum.   Hop onto the discord and maby someone there can help

bnxflazy, 2018-09-10:
I see, I haven't use discord thus far, thanks for the reminding though

bnxflazy, 2018-09-10:
which server should I use?

fennectech, 2018-09-10:
The one in this repo’s home page. It’s 5 am so i gotta go

CTCaer, 2018-09-12:
https://github.com/CTCaer/hekate/issues/109

closed 2018-09-12 by CTCaer