CTCaer/hekate #110: Error when restoring

chxk, 2018-09-14:
I have backup Boot 0/1 and Raw GPP in my exFAT TF card ,but it shows " Error (4) while opening backup....." .How can I solve this problem?

chxk, 2018-09-14:
I made it when I change the version to 3.2 and change some locations of backup files.(backup is done by version 4.0)

CTCaer, 2018-09-15:
Error 4 is `FR_NO_FILE /* (4) Could not find the file */`
You need to put the restore files into /backup/<emmc s/n>/restore/

You can create these folders by dumping sth, like fuses. Or manually by reading your emmc info.

closed 2018-09-19 by CTCaer

GiovanH, 2018-11-23:
Because this is marked as Informative, and I had the same issue, I wanted to note this:

Some tools dump boot0 and boot1 as boot0.bin and boot1.bin. **Hekate requires the file to be named BOOTX with no extension.**