CTCaer/hekate #111: hekate v4.1 release info

CTCaer, 2018-09-18:
Just want to say that the release is ready weeks now.

It has full 6.0.0 support (hekate/atmo patches) and it will be released after making sure that the "leaked" 6.0.0 with revision 4.0 is indeed what it's going to be released officially.

Even if that's the case, don't update. Not until all new logging is RE'ed.

CTCaer, 2018-09-19:
[image]

mbarisa, 2018-09-19:
any ETA when you are going to release 4.1?

mbarisa, 2018-09-19:
Great, thanks. :+1:

IFireflyl, 2018-09-19:
Just to confirm, if I went completely stupid and updated to 6.0.0 last night I should still be able to homebrew the Switch once 4.1 is released sometime in the next few days? Or should I wait a while to make sure there aren't other issues that need to be addressed? Pretty much the only thing I want to do is backup my game saves. 😁

CTCaer, 2018-09-19:
Well so I had time to check it. I'm disappointed with ninty..

mbarisa, 2018-09-19:
Awesome, what is ninty?

CTCaer, 2018-09-19:
nintendo

CTCaer, 2018-09-19:
https://github.com/CTCaer/hekate/releases/tag/v4.1

closed 2018-09-19 by CTCaer