CTCaer/hekate #114: AutoRCM proper shutdown method not optimal for people using PCs for payloads

noahc3, 2018-09-19:
Hekate 4.1 introduced a feature where if the payload is launched after shutting down, instead of entering the menu, it will shutdown the switch properly. This is great for modchips (and dongles depending on who you ask) but not very helpful and even sometimes annoying for people who use a PC for payload launching. The main problem is many people only ever shutdown their Switch when they want to swap files on their SD. Having to push hekate twice to boot again after doing this is mildly annoying.

It isn't a huge issue, but having an option to disable this functionality in settings would be nice. Perhaps also including a message about what is actually happening on the image that displays would also be nice so people aren't confused.

parkerlreed, 2018-09-20:
This actually explains a lot... It would show the hekate boot splash and then go dark. I was confused as to why this was happening. Now I know.

CTCaer, 2018-09-20:
> Auto full power off when the device woke up from HOS' power off
Usefull with modchips/dongles when using AutoRCM. **(You can see it as a breathing backlight with hekate's logo)**

That's true though. I have a different workflow (the one below), so I didn't think a lot of it.

Until I add an option to disable it, for now do the following:
`When you power off from HOS, while having the jig, press VOL UP + PWR to enter rcm.`
`With autorcm, just PWR.`
This will register the buttons as a wake up reason and it will skip this.

CTCaer, 2018-09-24:
https://github.com/CTCaer/hekate/commit/381d5c9236ed17270ddae4a2b181c165a2673d1e

closed 2018-09-24 by CTCaer