CTCaer/hekate #116: Black screen after clicking CFW Hekate 4.1

zeq2mapper, 2018-09-20:
I’m On 4.1.0. I inject the payload fine. The menu comes up fine. When I goto launch the CFW. It gives me writing and a black screen. Eventually booting normally. 

What and I doing wrong? Is it a bad SD a card?

CTCaer, 2018-09-20:
You are in 6.0.0?

zeq2mapper, 2018-09-20:
I’m on firmware 4.1.0

CTCaer, 2018-09-20:
Oh wait a minute.
You said writing and a black screen.
Actually this is how it boots. Writing is what it currently does, and then it switches off the display and boots in HOS.

So where's the problem?

zeq2mapper, 2018-09-20:
I don’t see any splash or anything. It just boots into regular mode. After a while it crashes and gives me this error 2162-0002

zeq2mapper, 2018-09-20:
After booting. I went to Albums. And it loads Albums as normal. Why would this be happening??

CTCaer, 2018-09-20:
That's a problem with your sd files. (modules/kips/etc)

Also the bootlogo is only shown in autoboot (except if you mean the nintendo/switch logo).

zeq2mapper, 2018-09-20:
So maybe I need to extract the SDfiles again to my SD card? I’ll try that again then

zeq2mapper, 2018-09-20:
The same thing happened again. It just loads OFW like normal. The Nintendo logo comes up then it quickly come off. It takes 3-5 seconds for the other sketch logo then it boots like normal. Then it crashes with that code I told you above. 

What could be causing that??

zeq2mapper, 2018-09-20:
Do I need to just update to 6.0?

CTCaer, 2018-09-20:
The files you are using are probably incompatible with 4.1.0.

zeq2mapper, 2018-09-20:
The SD Files? How do I find the ones that are compatible?

tumGER, 2018-09-20:
https://github.com/tumGER/SDFilesSwitch/issues/97

I'll stamp this as a user fault - This is the only case of crashing I have received yet

CTCaer, 2018-09-24:
^

closed 2018-09-24 by CTCaer