CTCaer/hekate #117: KIP1 patches on SD card

misson20000, 2018-09-20:
Atmosphere's loader sysmodule has recently(-ish) implemented [exefs patching](https://github.com/Atmosphere-NX/Atmosphere/blob/master/stratosphere/loader/source/ldr_patcher.cpp) via IPS files stored on the SD card, in what I find to be a very convenient arrangement. Patch files indexed by an NSO's build-id, and are very easy to author as they are made between two copies of an uncompressed NSO.

It would be very nice to have a system like this for patching KIPs, rather than hardcoding each patch in the bootloader. Perhaps KIP patching would also be a good candidate for splitting into a new module.

misson20000, 2018-09-20:
This may be a duplicate of #60.

CTCaer, 2018-09-20:
It is and it isn't a duplicate.

Note that I don't like IPS patching at all. I prefer writing a custom and simple format which can be like what hekate currently does, with 2 simple records: u32 and array patching.

I already have a working PoC internally, both hos/pkg1/pkg2 as a module.

But until I finish my GUI I wont release it.
(And I don't want to lose time on implementing just the kip patching from sd for now.)

CTCaer, 2019-06-30:
https://github.com/CTCaer/hekate/commit/ba0f29bc5caaf4e045bfac1f139f7f9990318db0

closed 2019-06-30 by CTCaer