CTCaer/hekate #118: BRICKED.

1Canna, 2018-09-20:
Hi there when using a knwon working sd card, hekate 4.1 bin and latest sd card files..

On a untouched never hacked switch. i have tried to backup my nand today boot0/1 backup went fine..
raw gpt ?rawnand.bin backup froze at 99% dumping for over a hour. the cancel dump button combination did not work i was force to hard reset...
Now the switch will not boot hekate payloads at all versions 3.2 4. 4.1
the switch will boot other payloads memdump biskeys etc..

After the hard reset of hekate freezing at 99% dump i restarted switch into normal mode to try boot stock horizon the switch will not boot at all.. the dumping process has bricked or stopped my switch booting normally or hekate payload .... so now what do i do ? why did this happen if its just a simple dump ?

1Canna, 2018-09-20:
rcm works fine... all other payloads work fine.. hekate payloads dont work, but then when i load memloader mount emmc, then turn off and used hekate 3.2 payload just then it worked,,but cannot load hekate any other way again now.. its so weird.. and erm yes i only see the first logo the smaller nintendo logo... i have tried with and without sd card reader and by disconnecting battery.. this all happend when it froze at 99% dumping nand.

CTCaer, 2018-09-20:
[iplctc1.zip](https://github.com/CTCaer/hekate/files/2402286/iplctc1.zip)
try this and don't boot yet.
Take a picture of eMMC info and sd card info. Then try to boot stock.

1Canna, 2018-09-20:
payload does not work...

CTCaer, 2018-09-20:
[iplctc3.zip](https://github.com/CTCaer/hekate/files/2402368/iplctc3.zip)
this?

1Canna, 2018-09-20:
this payload worked i have included emmc dump sd card was not inserted
[This payload worked.zip](https://github.com/CTCaer/hekate/files/2402374/This.payload.worked.zip)

1Canna, 2018-09-20:
did not load instant after a minute the payload has loaded, but it freezes... Does nothing. 
BATTERY 0% CHARGE +)MA
battery was fine when i first used hekate 4.1 to backup 94%

but what about hekate 4.1 dumping till 99% then freezing then my console will not boot normally ?

CTCaer, 2018-09-20:
It seems that my questions do not show up on your side...
1. What did you test now? 3.2? iplctc3? 4?
2. What fw version you have...
3. Also what did you used to inject (and the complete batt, 0% and volts)?

I hope you don't have an empty battery...

1Canna, 2018-09-20:
Ok so i have disconnected the battery and launched the payloads they are all working now. but the problem is still that when i dumped raw emmc gpt with hekate 4.1 stuck at 99% dumping for a hour i had to force power off, and then my switch would not boot, when it froze it had 94% battery..

CTCaer, 2018-09-20:
You keep not answering my questions.. and continue to reply with what you did and not what I asked you to do.

1Canna, 2018-09-20:
i have just mentioned i have tested the payloads you sent me,, and they did not work, now i have removed the battery the payloads work... i have mentioned im on 5.1.0 fw.
and when battery is installed it freezes and shows 0% ) charge.. when the battery is disconnected it shows 0%

CTCaer, 2018-09-20:
Run the battery desync.
It seems that the problem is at your usb charging IC (which ofc no bootloader/cfw touches, there's not even a dormant code for these).

Can you manage to run the battery info?

1Canna, 2018-09-20:
Ok so battery installed again, i load hekate 4.1 it takes really long to load.. once it loads it freezes and shows battery 100% 4146mv  and charge +123ma.. switch is now frozen

i am unable to run bat desync as the switch is frozen on the hekate menu...

But the switch was working fine before i used hekate 4.1

i have just checked the switch and the battery is now 0.0% 4140mv charge +81ma still frozen

CTCaer, 2018-09-20:
That's a power issue.
And it is caused by the RHOM BM92T30MWV usb-pd charger ic and a bad charge from the charger or wall socket to the switch.

1Canna, 2018-09-20:
it was not connected to charger, it has been connected to pc...  just to dump nand with hekate 4.1 as soon as it reached 99% it froze and would not do anything had to hard reset.. then the switch would not boot normally

fennectech, 2018-09-20:
Hmm.   Perhaps your PC try’s to act as a charger?

1Canna, 2018-09-20:
the problem lies with the switch and the hekate 4.1 which i tried to make a backup of nand earlier and froze at 99%.. switch was not even connected when i was in hekate

fennectech, 2018-09-20:
I dont know mate.   But. Ctcaer says its the power IC.  Wich is the charger brick

Rickardo987, 2018-09-20:
Find a fast charge compatible port. It should supply enough voltage to fast charge the switch.

CTCaer, 2018-09-20:
connected to pc is not "not even connected".
It seems that this can eventually happen. I have my switch since march 3 2017 and with all the abuse it took, it still works like new. 
And I always use nintendo's cable from the charging grip.

So, 0%->100%->0% is either battery or usb-pd failure.

This switch as you said is new and untouched. So you can even ask for a DOA replacement.

1Canna, 2018-09-20:
the switch does not need charging. i know that for a fact.. but i just noticed in hacdiskmount that safe partition will not mount, so am thinking nand corruption.

CTCaer, 2018-09-20:
Well backup does not write anything to emmc. Also emmc is not even needed for hekate to boot.
You can remove both emmc and sd card reader, and hekate will still boot.

Also no need for charge is different than able to provide power.
You can check on the datasheets how the 2 charger IC components work.
Basically, the battery never provides power to switch. It provides power to these and these transfer the power to the pcb.

fennectech, 2018-09-20:
GPTrestore repairs the gpt partition information

1Canna, 2018-09-22:
i appreciate the responses , im not trying to blame hekate.. im just looking for a solution.. 
So gpt restore does not wipe the contents ? just changes partition info ?
thank you all for your help im having a pro look at the mainboard test the components for me..
Just my luck... will inform here once switch is working again thank you

CTCaer, 2018-09-22:
Why you want to break it apart?
You don't have a warranty?

Don't use gpt restore, it has nothing to do with your case.

CTCaer, 2018-09-25:
Closing this as invalid.
If you find that sth is not right, pls feel free to re-open this.

Note: modifying hekate to configure the 2 charger IC, may help you find the problem in that matter.
I still believe it's best to use your warranty before voiding it completely...

closed 2018-09-25 by CTCaer

1Canna, 2018-09-27:
i would like to apologise for any confusion or misunderstanding. i had said it was due to hekate at the time as thats all i used.. and never had issues before. i guess i just got unlucky at the wrong moment. during the backup process, damn thing backup died at 99% lol.
 i have been told by a electronic expert that they think the battery ic chip has gone bad would this stop ofw or hekate loading correctly ??.. As i know a simple backup should not cause issues or write any corruption to nand it was just odd at the time... so case closed hopefully problem has been found and will be fixed

CTCaer, 2018-09-27:
Check your cable also and try not to use it to your good switch.

disalinas09, 2019-08-25:
Hello, I think it has to do with the NAND chip, I was using my Nintendo normally but wanted to return to a previous state, I put the Nintendo in RCM mode and restore the backup I had previously done, and the Nintendo was off, stopped entering RCM mode and did not load or give signs of life, buy another NAND eMMC chip, I opened the console and put the new NAND chip, this factory chip comes empty, you have to load some restore, by placing it connect the console and enter RCM mode normally, the battery charge marked me 95%, so it was not a charging problem or anything to do with that, the nand chip was damaged by the restore and the console moria, switch back to the previous NAND memory to corroborate this and it was like this, when making the change stopped entering RCM mode, there was no way, thinking that the previous memory had been damaged because at some point restore failure, put the new NAND chip, put the console in RCM mode and make the restore again, and what happened?, again I stop booting, this way I damage both memories, and I do not intend to buy another, for this reason I am in this post, I need to know how I can do to boot the console RCM mode or whatever to recover the memories.

Translated with www.DeepL.com/Translator

CTCaer, 2019-08-29:
@disalinas09 
Your eMMC backup is obviously corrupted or missing files.
The nand chips do not die like that. You should not waste money on that.

You need to check your eMMC backups first and then try to restore a working one.