CTCaer/hekate #119: sleep mode

novalkun, 2018-09-20:
i cannot sleep the switch with 4.1 release. is it normal?

ErichMoraga, 2018-09-20:
Did you transfer the extra file?

ErichMoraga, 2018-09-20:
libsys_lp0.bso as seen here...
https://github.com/CTCaer/hekate/blob/master/README.md

novalkun, 2018-09-20:
i have that libsys_lp0.bso directly from hekate sdfiles 9.0.1. but it's not working

CTCaer, 2018-09-20:
v4 works?
Also do you see any warning about sleep mode in main screen?

novalkun, 2018-09-20:
v4 works fine.

yes. when i click the power touch button, it says some warning before my switch goes to sleep. and then i choose sleep. after i click it, my switch goes to black screen, but when i press any button, the switch does not wake up. 
i test with SX OS, to see if anything wrong with the 6.0 firmware, but my switch can sleep and wake up fine. 

i do not know if it is a sleep problem or a wake up problem

CTCaer, 2018-09-20:
Which ini and boot entry you used?
In Stock through hekate it works?

novalkun, 2018-09-20:
I choose CFW boot option.

I have not used ctcaer 4.0 for 6.0 firmware yet. I'll try to run that later.

CTCaer, 2018-09-20:
Don't try hekate v4.0.
Try the `Stock` option

CTCaer, 2018-09-20:
It's not exactly stock. It's stock as far as kips go.
It still goes through hekate (hw config) and hekate patches secmon and loads sleep mode (lp0) config.

novalkun, 2018-09-21:
i have tried stock option. sleep mode works (yes). why cfw option is not working T_T

novalkun, 2018-09-21:
ah i am sorry. think the option for using CFW+sys-netcheat or stock_sysnetcheat have made the sleep mode broken. if i am using only CFW or stock, it's fine

Reisyukaku, 2018-09-22:
Guess I'll mention this here, but on 6.0, the sleepmode issue reported on reinx exists on hekate too. It's the debug svc kernel patch. I had someone in my server test.

novalkun, 2018-09-22:
yes. it seems recent debugmode is not compatible to 6.0. thanks. i am waiting for good news

CTCaer, 2018-09-22:
It is compatible with 6. It's just that secmon added checks for debug mode.
The thing is that debugmode and full svc perms are not supposed to be used all the time.
Dunno how much of a wanted fix that is.

Reisyukaku, 2018-09-23:
Ah, didn't know about the checks. That explains why atmo is fine. I'll put that at the end of my list since i dont know many people that use it lol.

plutooo, 2018-12-09:
will you fix this?

plutooo, 2018-12-11:
ping

CTCaer, 2018-12-12:
@plutooo 
I plan to.
For now I haven't found why cpu suspend is not called and the best result/reaction I got was a light yellow panic screen.
Additionally, enabling debug mode via secmon and not kernel produces the same problem ("black screen w/o backlight").

SciresM, 2019-01-22:
It's because the kernel will try to output debug stuff over UART, and UART is not set up. Exosphere currently fixes this (and thus ams has debugmode on by default), but you can probably just set uart up yourself before handover to tz.

Relevant code: 

https://github.com/Atmosphere-NX/Atmosphere/blob/e0f1e637f79a6eae88178a6576a3a74a953d1929/exosphere/src/bootup.c#L218-L220

https://github.com/Atmosphere-NX/Atmosphere/blob/e0f1e637f79a6eae88178a6576a3a74a953d1929/exosphere/src/warmboot_main.c#L59-L64

CTCaer, 2019-01-22:
@SciresM 
There were some jokes about this ("inb4 it's uart that hangs it"), even if it made sense because usage of soc parts with clock off hang the SoC.

The thing is that I tried to enable UARTA via hekate before 2 months when I saw that commit.
But it was always failing (maybe sth wrong with the old and new uart init).

I also tried stubbing functions that use it before and most of the times got hangs and sometimes a panic (probably missed some).

Anyway, now that I have your confirmation, I can try again and focus only to uart.

SciresM, 2019-01-23:
@CTCaer yeah, when I added those commits all debugmode issues related to sleep went away for ams, so I'm pretty confident.

CTCaer, 2019-02-11:
https://github.com/CTCaer/hekate/commit/b3f8961e45d702164dc0f2e0325d6cee90683800

closed 2019-02-11 by CTCaer