CTCaer/hekate #120: Partitial raw nand backup verification always fail at 99%

Ch0cola, 2018-09-21:
When I try to perform a full backup of my nand on a 16GB microSD card, always show a error message at 99%:

> SD Card and eMMC data (@LBA 03A3E000), do not match!
> Verification failed..
> 
> Press any key and try again...

If I try to retry, same message appear at 99%

The nand backup after join the 15 parts seems correct, I can open and browse it with HacDiskMount using my BIS keys without problems, I restore my Switch with this backup without problems too.

Switch firmware: 5.1.0
Hekate version: 4.0

CTCaer, 2018-09-21:
I thought I fixed that:
https://github.com/CTCaer/hekate/issues/102

What's your backup size?
Also how you made the backup? 7parts + 7parts + last part? or differently?
I need to see if it's the same pattern.

Ch0cola, 2018-09-21:
The raw nand size are 30,0 GB (32.212.254.720 bytes) on Windows file properties.

Yes, the backup pattern are 7 + 7 + 1 because the size of microSD card.

And yes, I use hekate v4.0 because didn't see the recent version, now I update to v4.1, you need the backup report with the recent version?

CTCaer, 2018-09-21:
Can you try with hekate 4.1?

This is fixed in that version.

Ch0cola, 2018-09-21:
At this time can't try, but when I'm arrived to home I test it and say you the report.

Sorry for the apologies, didn't see that new version are arrived.

Ch0cola, 2019-04-21:
Any way to restore the 30GB nand?

My last clean OFW backup are extracted before the fix and now I can't restore it because say that size are not equal to eMMC.

CTCaer, 2019-04-22:
Yes, just trim it.
There are tools that let you split the file into 2. Choose 31,268,536,320 bytes

Ch0cola, 2019-04-22:
A lot of thanks @CTCaer, I think on trimm the unused bytes of the file but when I see the Switch on this state I enter on panic and I ask and wait for the response before do any more steps xD

My Switch live again, a lot of thanks ^^

-------------------------------------------------

For the people that have the same issue with his backup, I trimm the file through linux terminal but one friend on Windows use this: https://www.mynikko.com/trimmer/

Correct size for one unic file: 31,268,536,320 bytes
Correct size for the last part (rawnand.bin.14): 1,203,765,248 bytes