CTCaer/hekate #123: sxos

nitrozz, 2018-09-25:
sxos load but no config menu 
hetake 4.2 and sxos 1.9

ErichMoraga, 2018-09-25:
It works fine, but you should also consider getting SX OS v2.

nitrozz, 2018-09-25:
sxos option/menu dont show as usual when booting sxos

ErichMoraga, 2018-09-25:
I thought that was normal once it sees your license.dat file.  Hint... If you really want to get to that menu right after loading the payload, temporarily rename the license ;-)

ErichMoraga, 2018-09-25:
I think you can also get to it by holding down Vol + while the payload is loading.

CTCaer, 2018-09-25:
Please read the manual. Here's not a support forum on how to enter the boot menu.

For any payload that is launched, the keystroke must happen the moment you press PWR to launch it.

Rickardo987, 2018-09-25:
On that note maybe you could add a second delay between pressing “POWER” to launch the payload and actually launching it so people have a moment to press any config key (such as “VOL +”)