CTCaer/hekate #124: Add .sh to the linux/macOS files in the linker scripts folder.

noirscape, 2018-09-26:
This will make linux/macOS recognize these files as scripts, and in the case of macOS makes them executable by double clicking on them.

As these files are part of the release, I can't PR this change to the repository.

CTCaer, 2018-09-26:
you don't use ./script anyway?
or you double click it?

noirscape, 2018-09-26:
@CTCaer - On mac, files without an extension aren't executable as terminal scripts unless they have the .sh extension.

For that matter, the files also don't have the executable bit set, meaning you can't execute them either through the terminal unless you change permissions.

fennectech, 2018-10-02:
On linux you also cannot execute files without the executable bit sans windows files with wine

ThatNerdyPikachu, 2018-10-03:
But if someone uses Linux on a daily basis, they should know how to fix this issue

noirscape, 2018-10-04:
> Most definitely. “chmod +x file.sh”. It takes seconds to type.

It being "easy to fix" isn't an excuse not to fix it. It's a 5 second fix,
which at least would highly simplify merging NAND dumps on macOS (terminal
scripts that are identifiable as terminal scripts by extension run when you
double click them).

noirscape, 2018-10-04:
@fennectech - That's why I made an issue. They aren't in the repository, otherwise I would've made the PR right now.

fennectech, 2018-10-04:
Well when they are committed we can fix them if he doesn’t.  Thats the benefit of open source.  You can actually fix them yourself.

CTCaer, 2018-12-06:
New files have the .sh ext

closed 2018-12-06 by CTCaer